verilator/verilator #4297: Suspendable procedures never split into multiple files

Jomit626, 2023-06-15:
I have a testbench that calls a lot of tasks with timing controls in a initial block. Verilator generates a 9MB+ cpp file. It takes forever to compile. Here is a testcase that also generate a large cpp file (2.7MB).
[t_coroutine_split.tar.gz](https://github.com/verilator/verilator/files/11760393/t_coroutine_split.tar.gz)

In V3Sched [`orderSequentially`](https://github.com/verilator/verilator/blob/d1ff81a41c75ae4568bcb6d4f8f0cbc5688e76a1/src/V3Sched.cpp#L214), a suspendable procedure will be convert to a separate function and be replaced with a function call statement. But [`splitCheck`](https://github.com/verilator/verilator/blob/d1ff81a41c75ae4568bcb6d4f8f0cbc5688e76a1/src/V3Sched.cpp#L122) is never used to check that new function.

I've tried simply adding `splitCheck` in orderSequentially:
```diff
@@ -254,8 +254,11 @@ void orderSequentially(AstCFunc* funcp, const LogicByScope& lbs) {
                             bodyp
                                 = new AstWhile{flp, new AstConst{flp, AstConst::BitTrue{}}, bodyp};
                         }
+                        subFuncp->addStmtsp(bodyp);
+                        splitCheck(subFuncp);
+                    } else {
+                      subFuncp->addStmtsp(bodyp);
                     }
-                    subFuncp->addStmtsp(bodyp);
                 }
             } else {
                 logicp->unlinkFrBack();
```

Verilated code splits into smaller files, but there are errors during compilation.

``` log
Vt_coroutine_split___024root__DepSet_h18267415__0.cpp: In function ‘VlCoroutine Vt_coroutine_split___024root___eval_initial__TOP__0(Vt_coroutine_split___024root*)’:
Vt_coroutine_split___024root__DepSet_h18267415__0.cpp:41:1: warning: no return statement in function returning non-void [-Wreturn-type]
   41 | }
      | ^
Vt_coroutine_split___024root__DepSet_h18267415__0.cpp: In function ‘void Vt_coroutine_split___024root___eval_initial__TOP__0__0(Vt_coroutine_split___024root*)’:
Vt_coroutine_split___024root__DepSet_h18267415__0.cpp:48:5: error: unable to find the promise type for this coroutine
   48 |     co_await vlSelf->__VtrigSched_h7a97c1b5__0.trigger(0U,
      |     ^~~~~~~~
```
Any ideas?

wsnyder, 2023-06-17:
As the GCC error suggests I don't believe coroutines lend themselves to being split.  Maybe @kbieganski  has other suggestions.

Meanwhile, see if you can workaround by refactoring the Verilog code to something less expansive.  (E.g. your specific example could be a FSM using normal synthesis constructs (always @ negedgeedge clk).

kbieganski, 2023-06-17:
`splitCheck` simply doesn't know how to split coroutines. If the function we're splitting is a coroutine, this means it has some `co_await` statements (represented by `AstCAwait`). If these statements get extracted to other functions and replaced by simple calls, two issues arise:

1. Those new functions that end up with `co_await`s in their body return `void`, whereas they should return `VlCoroutine`. This causes the compile error about the missing promise type.
2. The calls to these functions are not `co_await`ed.
    - Now without any awaits, the top-level function is considered a normal function (not a coroutine) that just happens to return a `VlCoroutine`. There's no return statement, and that's fine for C++ coroutines, but this isn't a coroutine anymore, so we get a warning about that.
    - More importantly, this would cause incorrect runtime behavior. Because the call is not awaited on, any awaits (timing controls) in the sub-functions would not block the top-level function. Effectively they would work like `fork..join_none`.

The solution is fairly simple – if a new function has `co_await`s, change its `rtnType` to `VlCoroutine`, and wrap the call in a `co_await`.

Jomit626, 2023-06-19:
Thanks for the suggestions. I will make a pr after I learn more about C++ coroutines.

PR #4307 (2023-06-23): Fix splitting coroutines (#4297)
Fixes #4297
closed 2023-06-23 by wsnyder