verilator/verilator #4300: Cannot pass string into parameter with -G

fadialzammar, 2023-06-18:
Hi all, I'm still fairly new to Verilator so please let me know if I'm just doing something incorrectly!

I'm running Verilator 5.010 on Ubuntu and wanted to change a SystemVerilog parameter in my top-level file using the `-G` flag. This would select a file to be eventually read with `$readmemh`.
```sv
module wrapper #(
    parameter string MEM_FILE = ""
) (
    // ...
endmodule
```

However, when I try to verilate my module with `verilator -cc wrapper.sv -I$SRC/rtl -GMEM_FILE="$SRC/mem/test.mem`, I get the following output:

```
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: /
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: h
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: o
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: m
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: e
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: f
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: a
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: d
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: i
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: C
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: P
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: E
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: t
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: r
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: s
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: c
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: /home/fadi/CPE333/otter/src/rtl/wrapper.sv:30:15: Illegal character in decimal constant: .
   30 |     parameter MEM_FILE = ""
      |               ^~~~~~~~
%Error: Exiting due to 17 error(s)
```

It looks like this is coming from [V3Number.cpp:284](https://github.com/verilator/verilator/blob/master/src/V3Number.cpp#L284), where each character in the path is being checked to see if it is a valid decimal number. Obviously, I am not trying to input a decimal number, so I'm not sure why `V3Number` was called. It seems like this could be with an issue with the parser, given this comment in the linked file turned out to be wrong 😉: 
`// Likely impossible as parser prevents hitting it`

wsnyder, 2023-06-19:
The quotes are removed by the shell, so Verilator doesn't know it is a string, try this

         -GMEM_FILE="\"$SRC/mem/test.mem\""

fadialzammar, 2023-06-20:
That fixed it, thanks!

closed 2023-06-20 by fadialzammar