verilator/verilator #4301: When I want to export multiple tasks with the same name, the generated code is compiled incorrectly.

Superchao-ai, 2023-06-19:
When I want to export multiple tasks with the same name, the generated code is compiled incorrectly.I'm using the 5.012 tag version. I remember the previous version didn't report errors.
The following is an example:
```verilog
module sub1;
   reg  [127:0] data [7:0];

   export "DPI-C" task write_data;
   task write_data;
   input [2:0] iA;
   input [127:0] iD;
   begin
      data[iA] = iD;
   end
   endtask

endmodule

module sub2;
   reg  [31:0] data [7:0];

   export "DPI-C" task write_data;
   task write_data;
   input [2:0] iA;
   input [31:0] iD;
   begin
      data[iA] = iD;
   end
   endtask
   
endmodule

module top;

   sub1 u_sub1();
   sub2 u_sub2();

   initial begin
      $display("Hello World!");
      $finish;
   end

endmodule
```
The generated code is as follows:
```cpp
// Verilated -*- C++ -*-
// DESCRIPTION: Verilator output: Symbol table internal header
//
// Internal details; most calling programs do not need this header,
// unless using verilator public meta comments.

#ifndef VERILATED_VTOP__SYMS_H_
#define VERILATED_VTOP__SYMS_H_  // guard

#include "verilated.h"

// INCLUDE MODEL CLASS

#include "Vtop.h"

// INCLUDE MODULE CLASSES
#include "Vtop___024root.h"

// DPI TYPES for DPI Export callbacks (Internal use)
using Vtop__Vcb_write_data_t = void (*) (Vtop__Syms* __restrict vlSymsp, CData/*2:0*/ iA, IData/*31:0*/ iD);
using Vtop__Vcb_write_data_t = void (*) (Vtop__Syms* __restrict vlSymsp, CData/*2:0*/ iA, VlWide<4>/*127:0*/ iD);

// SYMS CLASS (contains all model state)
class Vtop__Syms final : public VerilatedSyms {
  public:
    // INTERNAL STATE
    Vtop* const __Vm_modelp;
    bool __Vm_didInit = false;

    // MODULE INSTANCE STATE
    Vtop___024root                 TOP;

    // SCOPE NAMES
    VerilatedScope __Vscope_top__u_sub1;
    VerilatedScope __Vscope_top__u_sub2;

    // CONSTRUCTORS
    Vtop__Syms(VerilatedContext* contextp, const char* namep, Vtop* modelp);
    ~Vtop__Syms();

    // METHODS
    const char* name() { return TOP.name(); }
} VL_ATTR_ALIGNED(VL_CACHE_LINE_BYTES);

#endif  // guard

```

The compile error is as follows:
```log
Vtop__Syms.h:21:7: error: conflicting declaration ‘using Vtop__Vcb_write_data_t = void (*)(class Vtop__Syms*, using CData = uint8_t, struct VlWide<4>)’
   21 | using Vtop__Vcb_write_data_t = void (*) (Vtop__Syms* __restrict vlSymsp, CData/*2:0*/ iA, VlWide<4>/*127:0*/ iD);
      |       ^~~~~~~~~~~~~~~~~~~~~~
Vtop__Syms.h:20:7: note: previous declaration as ‘using Vtop__Vcb_write_data_t = void (*)(class Vtop__Syms*, using CData = uint8_t, using IData = uint32_t)’
   20 | using Vtop__Vcb_write_data_t = void (*) (Vtop__Syms* __restrict vlSymsp, CData/*2:0*/ iA, IData/*31:0*/ iD);
      |       ^~~~~~~~~~~~~~~~~~~~~~
```

wsnyder, 2023-06-19:
The way the IEEE standard is written DPI import/exports are completely global.  So you need to use two names.

There probably should be a verilation-time error if you do this instead of relying on the C compiler. I'll leave this issue open to have that added.  Perhaps you'd be willing to make a pull request to add such an error?

Superchao-ai, 2023-06-20:
I'm a beginner. It's difficult for me to modify verilator.

Superchao-ai, 2023-06-20:
After trying, "export" changed to the following, the generated code seems to be compiled without errors and appears to be functional.👍 
```verilog
module sub1;
   reg  [127:0] data [7:0];

   export "DPI-C" sub1_write_data = task write_data;
   task write_data;
   input [2:0] iA;
   input [127:0] iD;
   begin
      data[iA] = iD;
   end
   endtask

endmodule

module sub2;
   reg  [31:0] data [7:0];

   export "DPI-C" sub2_write_data = task write_data;
   task write_data;
   input [2:0] iA;
   input [31:0] iD;
   begin
      data[iA] = iD;
   end
   endtask
   
endmodule

module top;

   sub1 u_sub1();
   sub2 u_sub2();

   initial begin
      $display("Hello World!");
      $finish;
   end

endmodule
```

Thank you for providing solutions : )

wsnyder, 2023-06-20:
Glad that worked.  I found there was already an error for duplications, but it wasn't triggering due to the exports being similar in all but width.

closed 2023-06-20 by wsnyder