verilator/verilator #4309: SystemC port width doesn't match SystemVerilog port width

3N4N, 2023-06-22:
Hi, new to Verilator (and Verilog in general) so this might be a FAQ, but Verilator is generating the SystemC model `Vcounter` from my SystemVerilog mode `counter` with mismatched port width.  The width of `count_o` in `counter.sv` is 8 bits, but in `Vcounter.h` it is 32 bits.  Is this expected?  Any way to force Verilator to keep the exact number of bits defined in the SystemVerilog model?

```
module counter #(
    parameter int WIDTH = 8
) (
    input  logic             arst_ni,
    input  logic             clk_i,
    input  logic             up_i,
    output logic [WIDTH-1:0] count_o
);

  always_ff @(posedge clk_i or negedge arst_ni) begin
    if (~arst_ni) begin
      count_o <= 3;
    end else begin
      if (up_i) begin
        count_o <= count_o + 1;
      end else begin
        count_o <= count_o - 1;
      end
    end
  end

  function automatic logic [WIDTH-1:0] read();
    return count_o;
  endfunction

endmodule
```

3N4N, 2023-06-22:
I just read about `/* verilator sc_bv */` directive:

```
    /*verilator sc_bv*/
        Used after a port declaration. It sets the port to be of sc_bv<1width1>
        type, instead of bool, vluint32_t or vluint64_t. This may be useful if
        the port width is parameterized and different of such modules
        interface a templated module (such as a transactor) or for other
        reasons. In general you should avoid using this attribute when not
        necessary as with increasing usage of sc_bv the performance decreases
        significantly.
```

I'm guessing `sc_bv` is avoided by default because it's performs worse than C++ types like uint32_t.  Is this correct?

wsnyder, 2023-06-22:
Right, sc_bv is much much slower.  Also historically it wasn't used except for large widths (due to performance) and the sc_bv annotation came later.

closed 2023-06-22 by wsnyder