verilator/verilator #4313: Support property definition with empty parentheses

Donlon, 2023-06-25:
Currently, property definition with empty parentheses is not supported. For example, consider the following module.

``` SystemVerilog
module test (
    input clk,
    input rst,
    input tvalid
);

    property my_property_1;
        @(posedge clk) disable iff (rst) tvalid;
    endproperty

    property my_property_2();
        @(posedge clk) disable iff (rst) tvalid;
    endproperty

    sva_1: assert property(my_property_1()) else $error("assertion failed");
    sva_2: assert property(my_property_2()) else $error("assertion failed");

endmodule : test
```

This would result in such error

```
$ verilator --cc test.sv
%Error: test.sv:11:28: syntax error, unexpected ')', expecting IDENTIFIER or randomize
   11 |     property my_property_2();
      |                            ^
%Error: Internal Error: test.sv:1:8: ../V3ParseSym.h:124: Symbols suggest ending PROPERTY 'my_property_2' but parser thinks ending MODULE 'test'
    1 | module test (
      |        ^~~~
```

As specified in *IEEE Std 1800-2017, Annex A.2.10 Assertion declarations*, a `property_declaration` can has none or multiple `property_port_item` in its parentheses, so hopping such syntax can be supported by Verilator.

wsnyder, 2023-06-25:
This might only need a simple fix to verilog.y, perhaps you'd consider a pull request along with improving one of the tests?

svenka3, 2023-06-26:
> This might only need a simple fix to verilog.y, perhaps you'd consider a pull request along with improving one of the tests?

I attempted and believe got it working. 

Added "empty" to property_port_list as below:
```yacc
property_port_list<nodep>:  // ==IEEE: property_port_list
                /* empty */                              { $$ = nullptr; }
        |       property_port_item                              { $$ = $1; }
        |       property_port_list ',' property_port_item       { $$ = addNextNull($1, $3); }
        ;

```

Should I update a test and PR?

Regards

wsnyder, 2023-06-26:
Yes, please make a pull request.  Also rename the rule name to property_port_listE the E means it has an empty.  Also fix the indent of the production to line up please (the part in {})

svenka3, 2023-06-27:
Thanks, Wilson for the guidance. I notice we already have a rule with that E suffix as below:

```yacc
property_port_listE<nodep>:  // IEEE: [ ( [ property_port_list ] ) ]
                /* empty */                         { $$ = nullptr; }
        |       '(' property_port_list ')'        { $$ = $2; }
        ;

property_port_list<nodep>:  // ==IEEE: property_port_list
               /* empty */                                      { $$ = nullptr; }
        |       property_port_item                              { $$ = $1; }
        |       property_port_list ',' property_port_item       { $$ = addNextNull($1, $3); }
        ;
```
So we have 2 "empty" - i.e. without "()" and with "()" - any suggested name for the second empty please?

Donlon, 2023-06-27:
Maybe adding an empty entry to the `property_port_listE` is better, which is more consistent to the spec and other items in the file, like [this one](https://github.com/verilator/verilator/blob/f9164ab0d2726064f9df7a1b052f59b6a0953d2e/src/verilog.y#L1437).

``` YACC
property_port_listE<nodep>:  // IEEE: [ ( [ property_port_list ] ) ]
                /* empty */                       { $$ = nullptr; }
        |       '(' ')'                           { $$ = nullptr; }
        |       '(' property_port_list ')'        { $$ = $2; }
        ;
```

I also update the regression test code and can open a PR later.

PR #4317 (2023-07-26): Support property declaration with empty parentheses (#4313)
fix #4313
closed 2023-07-26 by wsnyder