verilator/verilator #4314: Non-LRM $psprintf support

svenka3, 2023-06-25:
Thanks for taking the time to report this.

What would you like added/supported?

> Prior to LRM adding $sformatf support, tools had $psprintf - though non-LRM, most tools support it. Would you consider adding this support? 
 
May we assist you in trying to fix this in Verilator yourself?

> I can attempt

wsnyder, 2023-06-25:
Maybe. What prevents updating the code to the 2005 standard?

svenka3, 2023-06-25:
Legacy code, customers tend to "believe" if we have as minimal "fixes" as possible while trying production code on a new simulator such as Verilator (new - as in with SVTB support). This one sounds easy to do.

gezalore, 2023-06-25:
I would argue that if you are getting a free simulator it would be fair you fix basic things like this in your codebase. Adding all kinds of vaguely specified legacy features to the project increases maintenance burden. I appreciate this particular instance might be simple, but still basic questions remain like, what does `$psprintf` do exactly? If it's the same as `$sformatf`, fixing your code is trivial, if it isn't, then what is it's *precise* specification?

wsnyder, 2023-07-02:
It looks like this is supported by the 4 large commercial simulators, and was used by OVM.  If you want to make a pull request to support this I'd accept it.  I'm closing this issue though as since it is not part of IEEE and won't be added unless someone contributes it.

closed 2023-07-02 by wsnyder