verilator/verilator #4318: Class - vif reference throws CPP redeclaration error

svenka3, 2023-06-27:
Thanks for taking the time to report this.

```verilog
interface if_1;
  bit clk;
endinterface : if_1

class drv_c;
  virtual if_1 vif;

  virtual task run();
    @(posedge vif.clk);
   // Below line causes Verilator CPP error
  //  Commenting out below line works 
    @(posedge vif.clk);
  endtask : run
endclass : drv_c

module m;
  drv_c d_0;

  if_1 u_if_1 ();

  initial begin
    d_0 = new();
    d_0.vif = u_if_1;
    //d_0.run();
    $finish(2);
  end
endmodule : m

```

Log:
```log
Vclk_blk___024unit__03a__03adrv_c__Vclpkg__DepSet_h69446cda__0.cpp: In member function ‘virtual VlCoroutine Vclk_blk___024unit__03a__03adrv_c::__VnoInFunc_run(Vclk_blk__Syms*)’:
Vclk_blk___024unit__03a__03adrv_c__Vclpkg__DepSet_h69446cda__0.cpp:42:18: error: redeclaration of ‘CData __Vtrigprevexpr_he98b049f__0’
   42 |     CData/*0:0*/ __Vtrigprevexpr_he98b049f__0;
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vclk_blk___024unit__03a__03adrv_c__Vclpkg__DepSet_h69446cda__0.cpp:19:18: note: ‘CData __Vtrigprevexpr_he98b049f__0’ previously declared here
   19 |     CData/*0:0*/ __Vtrigprevexpr_he98b049f__0;
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [/home/srini/tools/vlator/git_vlator/verilator//include/verilated.mk:245: Vclk_blk__ALL.o] Error 1
make: Leaving directory '/home/srini/proj/CVC_Labs/vsv_labs_2014/lab5/run_dir/obj_dir'
%Error: make -C obj_dir -f Vclk_blk.mk -j 1 exited with 2
```

What I noticed is the first reference to "posedge vif.clk" works, the second one throws this error.

What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.013 devel rev v5.008-304-gdfbd711e1 (mod)

What OS and distribution are you using?

> Windows, WSL
May we assist you in trying to fix this in Verilator yourself?

> Yes, but my exposure to Verilator source code is minimal, will attempt.

wsnyder, 2023-07-02:
V3Timing makes the duplication.  @kbieganski should we should uniquify - or can we use the same var twice?

kbieganski, 2023-07-03:
I believe [this PR](https://github.com/verilator/verilator/pull/4320) fixes it. @wsnyder Can you take a look? I responded to your comments.

wsnyder, 2023-07-05:
@kbieganski I backed out the #4320 change as it was causing core dumps on my system running t/t_mailbox.pl.  This might well be exposing something else but I can't debug further at the moment, sorry. Here's a backtrace:

```
- V3Ast.cpp:1217:     Dumping obj_vlt/t_mailbox/Vt_mailbox_057_reorder.tree
- V3Timing.cpp:970:   timingAll:

Program received signal SIGSEGV, Segmentation fault.
AstNode::sameTreeIter (node1p=0x555557350a00, node2p=0x1, ignNext=ignNext@entry=false, gateOnly=gateOnly@entry=false) at ../V3Ast.h:103
103         constexpr operator en() const VL_MT_SAFE { return m_e; }
#0  AstNode::sameTreeIter (node1p=0x555557350a00, node2p=0x1, ignNext=ignNext@entry=false, gateOnly=gateOnly@entry=false) at ../V3Ast.h:103
#1  0x0000555555c90b43 in AstNode::sameTreeIter (node1p=0x55555735b080, node2p=0x55555735a160, ignNext=ignNext@entry=true, gateOnly=gateOnly@entry=f$
#2  0x00005555564439ac in AstNode::sameTree (node2p=<optimized out>, this=<optimized out>) at ../V3AstInlines.h:62
#3  std::equal_to<VNRef<AstNode> >::operator() (rb=..., ra=..., this=0x7fffffffc869) at ../V3Ast.h:2513
#4  std::__detail::_Hashtable_base<VNRef<AstNode>, std::pair<VNRef<AstNode> const, AstVarScope*>, std::__detail::_Select1st, std::equal_to<VNRef<Ast$
#5  std::_Hashtable<VNRef<AstNode>, std::pair<VNRef<AstNode> const, AstVarScope*>, std::allocator<std::pair<VNRef<AstNode> const, AstVarScope*> >, s$
#6  0x00005555564457c8 in std::_Hashtable<VNRef<AstNode>, std::pair<VNRef<AstNode> const, AstVarScope*>, std::allocator<std::pair<VNRef<AstNode> con$
#7  std::_Hashtable<VNRef<AstNode>, std::pair<VNRef<AstNode> const, AstVarScope*>, std::allocator<std::pair<VNRef<AstNode> const, AstVarScope*> >, s$
#8  0x0000555556445cf3 in std::_Hashtable<VNRef<AstNode>, std::pair<VNRef<AstNode> const, AstVarScope*>, std::allocator<std::pair<VNRef<AstNode> con$
#9  std::__cxx1998::unordered_map<VNRef<AstNode>, AstVarScope*, std::hash<VNRef<AstNode> >, std::equal_to<VNRef<AstNode> >, std::allocator<std::pair$
#10 std::__debug::unordered_map<VNRef<AstNode>, AstVarScope*, std::hash<VNRef<AstNode> >, std::equal_to<VNRef<AstNode> >, std::allocator<std::pair<V$
#11 SenExprBuilder::getCurr (this=this@entry=0x7fffffffc6f0, exprp=exprp@entry=0x55555735b080) at ../V3SenExprBuilder.h:73
#12 0x0000555556448aa1 in SenExprBuilder::createTerm(AstSenItem*)::{lambda()#1}::operator()() const (__closure=<synthetic pointer>) at ../V3SenExprB$
#13 SenExprBuilder::createTerm (this=0x7fffffffc6f0, senItemp=0x55555735b130) at ../V3SenExprBuilder.h:209
#14 0x000055555654fff3 in SenExprBuilder::build (senTreep=0x55555735b1e0, this=0x7fffffffc6f0) at ../V3SenExprBuilder.h:225
#15 TimingControlVisitor::visit (this=0x7fffffffcde0, nodep=0x55555735b290) at ../V3Timing.cpp:738
#16 0x0000555555c4d51e in AstEventControl::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:2737
#17 0x0000555555c90544 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:958
#18 0x0000555555c90616 in AstNode::iterateChildren (this=0x5555573583c0, v=...) at ../V3Ast.cpp:921
#19 0x0000555556541c28 in VNVisitor::iterateChildren (nodep=<optimized out>, this=<optimized out>) at ../V3Ast.h:2539
#20 TimingControlVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ../V3Timing.cpp:955
#21 0x0000555555c5d3e8 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:271
#22 0x0000555555c5d32b in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:262
#23 0x0000555555c5cc65 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:182
#24 0x0000555555c3e1da in AstIf::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:3486
#25 0x0000555555c90544 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:958
#26 0x0000555555c90627 in AstNode::iterateChildren (this=this@entry=0x555557328a80, v=...) at ../V3Ast.cpp:922
#27 0x000055555654881d in VNVisitor::iterateChildren (nodep=0x555557328a80, this=0x7fffffffcde0) at ../V3Ast.h:2539
#28 TimingControlVisitor::visit (this=0x7fffffffcde0, nodep=0x555557328a80) at ../V3Timing.cpp:627
#29 0x0000555555d63026 in AstCFunc::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:618
#30 0x0000555555c90544 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:958
#31 0x0000555555c90616 in AstNode::iterateChildren (this=this@entry=0x5555572d2a50, v=...) at ../V3Ast.cpp:921
#32 0x000055555654b32e in VNVisitor::iterateChildren (nodep=0x5555572d2a50, this=0x7fffffffcde0) at ../V3Ast.h:2539
#33 TimingControlVisitor::visit (this=0x7fffffffcde0, nodep=0x5555572d2a50) at ../V3Timing.cpp:579
#34 0x0000555555d6327c in AstScope::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:1372
#35 0x0000555555c90544 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:958
#36 0x0000555555c90616 in AstNode::iterateChildren (this=this@entry=0x5555571ff500, v=...) at ../V3Ast.cpp:921
#37 0x000055555654327a in VNVisitor::iterateChildren (nodep=0x5555571ff500, this=0x7fffffffcde0) at ../V3Ast.h:2539
#38 TimingControlVisitor::visit (this=0x7fffffffcde0, nodep=0x5555571ff500) at ../V3Timing.cpp:570
#39 0x0000555555c5c2d9 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:70
#40 0x0000555555d634a6 in AstClass::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:2155
[... cut ...]
#47 0x000055555655270c in VNVisitor::iterate (nodep=0x555557180000, this=0x7fffffffcde0) at ../V3Ast.h:2535
#48 TimingControlVisitor::TimingControlVisitor (this=0x7fffffffcde0, nodep=0x555557180000) at ../V3Timing.cpp:961
#49 0x00005555565415ae in V3Timing::timingAll (nodep=0x555557180000) at ../V3Timing.cpp:972
#50 0x0000555555c2469c in process () at ../V3Global.h:131
#51 0x0000555555c26f98 in verilate (argString=...) at ../Verilator.cpp:641
#52 0x0000555555c27bf4 in main (argc=28, argv=<optimized out>) at ../Verilator.cpp:763
A debugging session is active.

```

PR #4320 (2023-07-05): Fix multiple edge timing controls in class methods (#4318)
Multiple edge timing controls in class methods would cause compilation errors on the generated C++ code. This is because the `SenExprBuilder` used for these would get recreated per timing control, resulting in duplicate variable names. The fix is to have a single `SenExprBuilder` per scope.

Fixes #4318.
PR #4344 (2023-07-07): Fix multiple edge timing controls in class methods (#4318)
Multiple edge timing controls in class methods would cause compilation errors on the generated C++ code. This is because the `SenExprBuilder` used for these would get recreated per timing control, resulting in duplicate variable names. The fix is to have a single `SenExprBuilder` per scope.

Fixes #4318 (second attempt).
closed 2023-07-07 by wsnyder