verilator/verilator #4322: SV class vif sampling does not work

svenka3, 2023-06-27:
With some trial-and-error I have a tiny SV class based BFM driver working in Verilator. However, VIF signal sampling does not seem to work. Please consider below code:

```verilog
interface if_1;

  bit [7:0] rdata;

endinterface : if_1

class drv_c;
  virtual if_1 vif;

  virtual task run();
    #100;
    $display ("%0t rdata: 0x%0h", $time, vif.rdata);
    #100;
    $display ("%0t rdata: 0x%0h", $time, vif.rdata);
    #100;
  endtask : run
endclass : drv_c

module dut (output wire [7:0] rd_val);
  assign rd_val = 8'haa;
endmodule : dut 

module m;
  drv_c d_0;

  if_1 u_if_1 ();
  dut u_dut (.rd_val (u_if_1.rdata));

  initial begin
    d_0 = new();
    d_0.vif = u_if_1;
    //u_if_1.rdata = 10;
    d_0.run();
    $finish(2);
  end
endmodule : m

```

In the BFM class, vif.rdata is sampled/printed. A dummy DUT drives this to 0xaa and other tools print
```log
100 rdata: 0xaa
 200 rdata: 0xaa
```
But Verilator prints:
```log
verilator --binary -assert vif_rd.sv
./obj_dir/Vvif_rd
100 rdata: 0x0
200 rdata: 0x0
```

What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.013 devel rev v5.008-304-gdfbd711e1 (mod)

What OS and distribution are you using?
> Windows WSL

May we assist you in trying to fix this in Verilator yourself?

> Yes, please - will attempt, but am not a simulator developer, so may need more help!

wsnyder, 2023-07-02:
@kozdra perhaps you have some tips?

I added

```
  initial begin
     #1;
     $display("%x", u_if_1.rdata);
    d_0 = new();
```

And the value is correct, so it's related to the virtual (as expected).

kozdra, 2023-07-03:
The reason is the `dut` cell gets entirely optimized out, along with the `= 8'haa` assignment, during deadAllScoped phase.  Variable usage tracking would need to be broadened for the virtual interface case so that the `rdata` variable knows it is still in use; I thought I did some of that in the initial PR, but this issue shows it turned out insufficient.

wsnyder, 2025-04-05:
This got fixed somewhere, passes on master.  Committed test to confirm.

closed 2025-04-05 by wsnyder