verilator/verilator #4325: Clocking within DPI-C exported task.

seban90, 2023-06-29:
Hello sir.

while developing IP test environment for my team, some features look unsupported yet. So I wonder I could request you to support this.

My code is below,
`export "DPI-C" task verilog_clocking;
task run_clocking;
	@ (posedge `TB_MAINCLOCK);
endtask

task verilog_clocking;
	run_clocking;
endtask // verilog_clocking

`
I wanted to export this task as C function binding DPI-C and use it to make simulation time elapsed.
But Compiler put the compilation log with Error ::unsupported

%Error-UNSUPPORTED: /home/steven/Project/shared/soc_dv/testbench/lib/sv/dpi_lib.svh:16:2: Unsupported: Timing controls inside DPI-exported tasks
   16 |  @ (posedge clk);
      |  ^
                  
As you know, this code works well in VCS or incisiv (I wish i could use those tools, but I can't  for now)
Could you consider this issue?
Thank you.
StevenKim (SoC developer, Harman connected services inc.)

wsnyder, 2023-06-29:
I thought there was a similar issue already filed but I couldn't find it.

Support for DPI calling timing routines may be fairly complicated given how coroutines work.   I suspect this is something you will need to provide a pull request or contract with a third party to get supported, as it's not something under high demand to get working otherwise. @kbieganski can perhaps provide information on how to get started.

wsnyder, 2023-06-29:
P.S. Note also this particular example may perform badly even if it was supported, clocks should be created by the outside eval calls, I would suggest seeing if you can redo your application to not advance clocks over the DPI.

kbieganski, 2023-06-29:
Issue #4225 covers this.

See [this comment](https://github.com/verilator/verilator/issues/4225#issuecomment-1562738548) if you're interested in the technical details.

closed 2023-06-29 by wsnyder