verilator/verilator #4334: %Error-ASSIGNIN: ... Assigning to input/const variable:

ylevhari, 2023-07-03:
I am getting errors that say:

%Error-ASSIGNIN: /rdata/dub/verif/ylevhari/verilator_test/fixed_src/dwc_ddrphy_pll.v:899:17: Assigning to input/const variable: 'vp'
  899 |         .vp({vp,vp}),

This is an assignment to an inout signal, which is legal verilog, how can I override, fix this ?

wsnyder, 2023-07-03:
There are multiple restrictions on tristate but it is hard to know which you are hitting.  Can you please post a complete example showing the error?

ylevhari, 2023-07-04:
You are correct , a more  detailed  description: 

At the top level the signal is defined as inout, at the instantiated module, it is inout
vcs, and ncverilog allow this (I think it is within the standard)

module top_module (
    input   wire                byp_mode,
    input   wire                en_cal,
    input   wire                force_cal,
    input   wire          [5:0] freq_sel,
    input   wire                gd,
    input   wire                gear_shift,

    ........

	Inner module (instantiated module
	
	inner_module I32[1:0] (
        .gd({gd,gd}),
        .vp({vp,vp}),
        .a(lock_phase_sel_ln[1:0]),
        .z(lock_phase_sel_l[1:0])
    );
 )

The inner module decleration

module inner_module( a, z, gd, vp );
    input a;
    output z;
    inout gd;
    inout vp;

    inv_buf_module  inv_bf( .a(a), .z(z), .gd(gd), .vp(vp) );

endmodule

wsnyder, 2023-07-07:
Can you please provide an example that compiles and shows the error?

phelter, 2023-07-14:
I'm seeing this too.  The simplest example I can come up with is:

test_top - that is an Or gate with a tristate test-bus that uses transmission gates to select which signal to view.  This isn't really used but you can get the idea how a larger design would have multiple things attached to a testbus and then you can monitor a signal by selecting the appropriate enable pin.

test_top.v:
```verilog
`timescale 1ns / 1ns
module test_top (
   inout wire atb,
   input wire atb_en,
   input wire a0,
   input wire a1,
   output wire x,
)

assign x = a0 or a1;
xgate  xgate_inst(.a(a0), .b(atb), .en(atb_en));

endmodule
```

xgate.v:
```verilog
`timescale 1ns / 1ns
module xgate (
   inout wire a,
   inout wire b,
   input wire en,
);

assign b = (en) ? a : 1'bz;
endmodule
```

Because  the `a0` is identified at the test_top as an `input`, but the transmission gate has `inout` as the connection for `a0` - this inappropriately creates an ASSIGNIN error for `a0`.

I can see the value of checking `ASSIGNIN` - for the:
* `assign` statement
*  a declaration of a `wire` or `reg`
* connecting an `input` to an `input`,
But when connecting `inout` to other `input` - it should be a different classification of error or not checked so that these can be suppressed independently for verilog that uses tristate busses in this manner - which can cause a LOT of false positives.

closed 2026-03-29 by wsnyder