verilator/verilator #4351: Data error when fork-join_none and automatic variables are used in loop

zxw060810, 2023-07-09:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```Verilog
`timescale 1ns/1ps
task automatic run();

    for (int i=0; i<5; i++) begin
        fork
            begin
                #1ns;
                $display("%s (%0d) @%0t: i=%0d", `__FILE__, `__LINE__, $time, i);
            end
        join_none
    end
    #2ns;
    for (int i=0; i<5; i++) begin
        automatic int ii = i;
        fork
            begin
                #1ns;
                $display("%s (%0d) @%0t: i=%0d", `__FILE__, `__LINE__, $time, ii);
            end
        join_none
    end
    #2ns;
endtask: run

module test();
    initial begin
        run();
        $finish;
    end
endmodule
```
expected results (from VCS)：
```shell
test.sv (8) @1000: i=5
test.sv (8) @1000: i=5
test.sv (8) @1000: i=5
test.sv (8) @1000: i=5
test.sv (8) @1000: i=5
test.sv (18) @3000: i=0
test.sv (18) @3000: i=1
test.sv (18) @3000: i=2
test.sv (18) @3000: i=3
test.sv (18) @3000: i=4
```
Actual results:
```shell
test.sv (8) @1000: i=0
test.sv (8) @1000: i=1
test.sv (8) @1000: i=3
test.sv (8) @1000: i=4
test.sv (8) @1000: i=2
test.sv (18) @3000: i=4
test.sv (18) @3000: i=4
test.sv (18) @3000: i=4
test.sv (18) @3000: i=4
test.sv (18) @3000: i=4
```

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.006 2023-01-22 rev UNKNOWN.REV 
(installed by brew on macOS )

What OS and distribution are you using?
macOS 13.4.1 on Apple Silicon M2

May we assist you in trying to fix this in Verilator yourself?
I am a DV engineer with hardware background, instead of software background. Therefore I don't think I am able to fix it.

kboronski-ant, 2023-07-14:
Thanks for reporting!

The reason for the first 5 lines in the actual output is something I'm already aware of and is already being addresed in #4356. Basically the `i` variable got copied into the task generated for the process spawned by the fork block before the delay could occur. That's the only way Verilator can handle it for now, as `i` can go out-of-scope before the forked process resumes after the delay. Expect that to be fixed in the near future.

The reason for the second set of 5 lines being all `i=4` is not known to me yet, so you might've found two bugs with this example.

kboronski-ant, 2023-07-17:
The current master branch gives me the following output for the second loop:
```
test3.sv (18) @3000: i=0
test3.sv (18) @3000: i=2
test3.sv (18) @3000: i=1
test3.sv (18) @3000: i=3
test3.sv (18) @3000: i=4
```

While the order is different, this looks fine to me. From what I've seen, the System Verilog spec doesn't say anything how to order the resumed processes. The ordering of statements in sequential blocks is not described in terms of timing. While the processes that wait 1ns are spawned sequentially, they all start at the exact same simulation time and they all resume after 1ns as they should.

closed 2023-08-29 by kboronski-ant