verilator/verilator #4357: Warning message for CASEINCOMPLETE generated in 5.002 is not occurring in 5.012

calebofearth, 2023-07-12:
A Verilator regression workflow on https://github.com/chipsalliance/caliptra-rtl/pull/149 demonstrates this issue, and further discussion is available on that PR.

When running a SystemVerilog project through Verilator regression with version 5.002, we observed the warning message `%Warning-CASEINCOMPLETE` which was a legitimate warning about the syntax in [one code file](https://github.com/chipsalliance/caliptra-rtl/blob/dev-msft-20230710/src/soc_ifc/rtl/soc_ifc_boot_fsm.sv).
Log files are [here](https://github.com/chipsalliance/caliptra-rtl/actions/runs/5522685192/jobs/10072632150?pr=149).

Running the same code with Verilator 5.012 does not generate this warning message.

This is a useful warning message and should be generated; it seems that a bug in Verilator is masking the warning message.

wsnyder, 2023-07-12:
The file mentioned https://github.com/chipsalliance/caliptra-rtl/blob/dev-msft/src/soc_ifc/rtl/soc_ifc_boot_fsm.sv has a bunch of imports and other dependencies, can you please make a self contained test that shows in 5.002 and not 5.012?

Please first see #3745 perhaps that applies?

calebofearth, 2023-07-12:
@wsnyder Thanks for the reply, #3745 covers our situation exactly.

In my opinion, this is a valuable warning to keep in place, and I agree with your [original response](https://github.com/verilator/verilator/issues/3745#issuecomment-1312567705) to the question regarding synthesized/silicon behavior.

The [citation](https://github.com/verilator/verilator/issues/3745#issuecomment-1312577911) from IEEE 1800-2017 is incomplete in #3745 - it is true that the default case is not required for a unique case, but the immediately preceding paragraph in that specification states:

> If the case is qualified as priority or unique, the simulator shall issue a violation report if no case_item
> matches. **_A violation report may be issued at compile time if it is possible then to determine the violation_** (emphasis 
> added). If it is not possible to determine the violation at compile time, a violation report shall be issued during run
> time. If the case is qualified as unique0, the implementation shall not issue a violation report if no
> case_item matches.
>
> NOTE—By specifying unique or priority, it is not necessary to code a default case to trap unexpected case
values. 

In the case of an enumerated variable that does not cover the full range of it's underlying type (e.g. a `logic [2:0]` with fewer than 8 enumerated values) the violation may be determined at compile time, so this warning should be kept in order to catch potential issues in silicon.

I believe that the change implemented in #3745 is correct for a `unique0-case` but not for `unique-case`, based on the above quote.

wsnyder, 2023-08-12:
Contributions to fix this would be welcome!

wsnyder, 2026-11-12:
I looked into this. The "possible to determine the violation" means that the tool can determine that a value outside of the enum will hit the case statement. In your example this cannot be determined by Verilator at compile time so must be done at runtime.  This is done in current master.  Therefore closing.  If you/someone finds this issue and wants a new warning that requires a new optional warning beyond what CASEINCOMPLETE currently does and IEEE requires, that is reports on missing values even though the enum is complete and fires an e.g. CASEINCOMPLETEENUM (suggest a name) which can usually be done completely at compile time, please file a new issue.

closed 2026-11-12 by wsnyder