verilator/verilator #4365: $time always return 1

andreili, 2023-07-16:
Hi, all.
I use a Verilator to simulate my CPU core and get bug after migration to 5 version (5.008, from Gentoo repo's).
Verilog code:
`    function static real get_ts();
        real ts = $time;
        return ts / 1000.0;
    endfunction

    function static void print_head();
        $fwrite(f, "+----------+----------+----------+-------------------------------------------------------+\n");
        $fwrite(f, "| %8s | %8s | %8s | %-53s |\n", "Time", "PC", "Opcode", "Instruction/Event");
        $fwrite(f, "+----------+----------+----------+-------------------------------------------------------+\n");
        $fwrite(f, "|%8.3fns|%10s|%10s| %-53s |\n", get_ts(), "", "", "Trace started.");
    endfunction

    function static void print_event(input string str);
        $fwrite(f, "|%8.3fns|%10s|%10s| %-53s |\n", get_ts(), "",  "", str);
    endfunction`

C++ file have next function:
`double sc_time_stamp() { return 0.0d; }`
And on output file, I see always "1" values: https://pastebin.com/Xc7pMhGR
Options, passed to Verilator:
VERILATOR_FLAGS += -cc --exe -j 6 --threads 10
VERILATOR_FLAGS += -x-assign fast
VERILATOR_FLAGS += -Wall
VERILATOR_FLAGS += --trace-fst --trace-structs --trace-threads 1
VERILATOR_FLAGS += --assert
VERILATOR_FLAGS += --coverage
VERILATOR_FLAGS += --build --hierarchical
VERILATOR_FLAGS += -CFLAGS "-I../../../vrf/ -I../../../sim_common/common/ -DTOP_NAME=$* -lpthread" -DTO_SIM
VERILATOR_FLAGS += -LDFLAGS "-lpthread -ljsoncpp -lfmt"
VERILATOR_INPUT = +librescan +libext+.v+.sv+.vh+.svh ../../vrf/$*.sv -f ../input.vc ../../sim_common/common/tb.cpp ../../vrf/$*.cpp

wsnyder, 2023-07-16:
I take it you expected to get 0 since your $time is defined to return always zero?

       double sc_time_stamp() { return 0.0d; }

What is your whole main()?  Perhaps there is a +1 there?

andreili, 2023-07-16:
I try this code, without any changes on results:
`    uint64_t cur_ts;
    double sc_time_stamp()
    {
        return cur_ts * 1.d;
    }
    int on_step_cb(uint64_t time, TOP_CLASS* p_top)
    {
        cur_ts = time;
        if ((time % SIM_PULSE_DELTA) == 0)
        {
            printf("SIM: running, time %ld ticks, WB_ADDR=0x%08x\n", time, p_top->o_wb_addr);
        }
...
    }`

andreili, 2023-07-18:
I check current code with Verilator V4 (4.106) - and see a normally workink $time function:

+----------+----------+----------+-------------------------------------------------------+
|     Time |       PC |   Opcode | Instruction/Event                                     |
+----------+----------+----------+-------------------------------------------------------+
|   0.001ns|          |          | Trace started.                                        |
|   0.005ns|          |          | Reset asserted                                        |
|   0.015ns|          |          | Reset de-asserted                                     |
|   0.385ns|          |          | Reset asserted                                        |
|   0.445ns|0x000000|0x20006f  | jal r0, 0x2              r0: 0x0<=0x2                 |
|   0.525ns|0x000002|0x87b7    | lui r15, 0x8000          r15: 0xb1cea343<=0x8000      |
|   0.535ns|0x000004|0x78113   | addi r2, r15, 0          r2: 0xa9466d6c<=0x8000       |
|   0.545ns|0x000008|0x1517    | auipc r10, 0x1008        r10: 0x40032e40<=0x1008      |
|   0.555ns|0x00000c|0xa1050513| addi r10, r10, -1520     r10: 0x1008<=0xa18           |
|   0.565ns|0x000010|0x1597    | auipc r11, 0x1010        r11: 0x7fc01512<=0x1010      |
|   0.575ns|0x000014|0xa1058593| addi r11, r11, -1520     r11: 0x1010<=0xa20           |
|   0.585ns|0x000018|0xb55663  | bge r10, r11, 0x24                                    |
|   0.595ns|0x00001c|0x52023   | sw r0, 0(r10)                                         |
|          |          |          |                                    MemWr: 0xa18=0x0() |
|   0.605ns|0x000020|0x450513  | addi r10, r10, 4         r10: 0xa18<=0xa1c            |

wsnyder, 2023-07-18:
The best thing to do is use the new version 5 contextp->time method as shown in the latest examples, rather than relying on the legacy operations. Can you try that please?

andreili, 2023-07-21:
@wsnyder I found the error - on assignment, on SV code. No warnings, no errors on compilation rtime (with "-Wall").

First function always return 0.001, second - return correctvalue:

```
function static real get_ts1();
    real ts = $time;
    return ts / 1000.0;
endfunction

function static real get_ts2();
     real ts;
     ts = $time;
     return ts / 1000.0;
 endfunction

wsnyder, 2023-07-21:
Sorry I didn't notice this code issue previously but the new version of verilator fixed this to be correct.  If a function is static the variables default to static, so your assignment correctly would happen only once in the first function, but every call in the second.  Alternatively use "automatic real ts = $time".

I don't see how to do a lint warning as the "intent" is different in both functions.

closed 2023-07-21 by wsnyder