verilator/verilator #4370: Differences in timing between V4 and V5

shareefj, 2023-07-18:
Hi there,

I'm trying to port our testbenches from V4 to V5 and after fixing the first few simpler issues, I'm now seeing differences in timing of when internal signals are sampled that causes failures.  I'm not sure if this is a bug or something I'm doing wrong that we got away with before.  Or perhaps V5 is just inherently different now that there is support for timing...

I've attached two screen grabs showing the difference on a SWD interface.  The first one below is correct.  Sorry, it could have been a cleaner wave grab but the vector is the internal state of the debug port and you can see it changing as the first 1 is sampled.  In the failing waves, the state seems to change on the same edge that the 1 is driven.  The sort of issue you'd get in a standard simulator if not using a clocking block that delays the inputs.  It sorta feels like this kind of issue.

[image]

[image]

Our testbench uses the OpenTitan framework which calls registered extensions every clock cycle: https://github.com/lowRISC/opentitan/blob/master/hw/dv/verilator/simutil_verilator/cpp/verilator_sim_ctrl.cc#L372  So it looks like the clock is toggled, if it's high we call all extensions which drive the interfaces, then eval is called.  

Not sure it will help but an example of the extension that drives this interface is here:  https://gist.github.com/shareefj/80eb98e0af0c434973cd104a73610c9d

wsnyder, 2023-07-25:
Are you using --timing in V5?   There are many changes in scheduling between V4 and V5; generally V5 should be correct while V4 cut corners.  Being "correct" though does mean Verilator behavior may take different choices where the IEEE allows unpredictable ordering, so it's quite likely there's some code that V4 took one way, and V5 another (but still correctly).  It's of course also possible there's a bug.

We'll need a small selfcontained case to understand if you still think there's problems in Verilator, thanks.

wsnyder, 2023-08-12:
Did you resolve this, or have an example?

shareefj, 2023-08-14:
No, I didn't have time to look at this so I'll close it for now.

closed 2023-08-14 by shareefj