verilator/verilator #4371: No warning regarding mixed blocking and non-blocking assignments if the assignment is within an expression

RRozak, 2023-07-18:
Verilator doesn't throw warning if there are a blocking assignment and a non-blocking assignment to the same variable if the blocking assignment is within an expression.
Here is the example:
```systemverilog
module t (/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   typedef enum int {
                 E01 = 1,
                 ELARGE = 2
                 } my_t;
   my_t e;
   int    i_cast;

   always @ (posedge clk) begin
      e <= E01;
      i_cast <= (e = ELARGE);
      $display("%d", e);
   end
endmodule
```
The warning is correctly thrown if the line ` i_cast <= (e = ELARGE);` is replaced by `e = ELARGE;`.
The warning:
```
%Error-BLKANDNBLK: assign.v:20:9: Unsupported: Blocked and non-blocking assignments to same variable: 'e'
                                    : ... In instance t
   20 |    my_t e;
      |         ^
                   assign.v:25:9: ... Location of blocking assignment
   25 |       e = ELARGE;
      |         ^
                   assign.v:24:7: ... Location of nonblocking assignment
   24 |       e <= E01;
      |       ^
                   ... For error description see https://verilator.org/warn/BLKANDNBLK?v=5.013
%Error: Exiting due to 1 error(s)

```
The command is `verilator --binary <path>  -O0`. The `-O0` only disables optimizations, which allows to reproduce the problem on the smaller example.
I used verilator in version `Verilator 5.013 devel rev v5.012-56-g4de1b2267` on Debian 11.

wsnyder, 2023-07-18:
Probably some issue related to recursion into the inner statement.  Perhaps you can make a pull to fix it?  Thanks

RRozak, 2023-07-18:
Currently I am working on another problem with assignments within expressions:
```systemverilog
class Base;
   
endclass

class uvm_callback extends Base;
   function bit callback_mode;
     return 0;
   endfunction
endclass

module t;
   initial begin
      uvm_callback cb = new;
      Base b = new;
      bit a = $cast(cb, null) && cb.callback_mode();
      $display("%d", a);
   end
endmodule
```
The cast is replaced by the assignment and then Verilaor throws error `Function not underneath a statement`.

I hope that fix for that problem will fix also this issue.

closed 2023-08-07 by RRozak