verilator/verilator #4373: BLKANDBLK error on SV queue

shareefj, 2023-07-19:
Version: Verilator 5.012 2023-06-13 rev v5.012-1-g9fb17e532

Hi, I'm trying to reuse some DPI code that makes use of SV queues but am seeing an error reported.  Initially I realised that this probably isn't supported as it's not synthesisable but I found at least one other ticket with someone using a queue so thought I'd check.

```
%Error-BLKANDNBLK: src/idex_vip_swd_if_dpi_wrapper_0/src/rtl/swd_if_dpi_wrapper.sv:53:16: Unsupported: Blocked and non-blocking assignments to same variable: 'u_swd_if_dpi.swj_seq_data_queue'
                                                                                        : ... In instance chip_sim_tb
   53 |   bit [31:0]   swj_seq_data_queue  [$];
      |                ^~~~~~~~~~~~~~~~~~
                   src/idex_vip_swd_if_dpi_wrapper_0/src/rtl/swd_if_dpi_wrapper.sv:105:17: ... Location of blocking assignment
  105 |   function void grpc_server_swj_sequence_push( input bit [31:0] value );
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   src/idex_vip_swd_if_dpi_wrapper_0/src/rtl/swd_if_dpi_wrapper.sv:163:30: ... Location of nonblocking assignment
  163 |               swj_seq_data_r <= swj_seq_data_queue.pop_front();
      |                              ^~
%Error: Internal Error: src/idex_vip_swd_if_dpi_wrapper_0/src/rtl/swd_if_dpi_wrapper.sv:163:33: ../V3Delayed.cpp:564: <= on read+write method
                                                                                              : ... In instance chip_sim_tb
  163 |               swj_seq_data_r <= swj_seq_data_queue.pop_front();
      |                                 ^~~~~~~~~~~~~~~~~~
```

The SV itself is pretty simple.  I declare the queue, it gets pushed by a DPI call and then popped in an always block.

```systemverilog

  bit [31:0]   swj_seq_data_queue  [$];

  function void grpc_server_swj_sequence_push( input bit [31:0] value );
    swj_seq_data_queue.push_back(value);
  endfunction

  always_ff @(posedge clk or negedge rst_n) begin : swj_seq_write_data_fifo
    if (!rst_n) begin
      swj_seq_data_r       <= '0;
      swj_seq_data_valid_r <= '0;
    end else begin
      case(swj_seq_data_valid_r)
        1'b0 : begin
          if (swj_seq_data_queue.size() > 0) begin
            swj_seq_data_r       <= swj_seq_data_queue.pop_front();
            swj_seq_data_valid_r <= '1;
          end
        end
        <snip>
```

Any reason I can't do this?

kbieganski, 2023-07-19:
I think this is a false positive due to how `V3Delayed` detects writes to variables. It doesn't distinguish between the LHS and RHS and just [sees a read/write reference](https://github.com/antmicro/verilator-1/blob/5a694ccb8a73af8189b4ca599c1ebedfbf27ca4d/src/V3Delayed.cpp#L560) to the queue in the assignment. It just assumes that it's on the LHS, which is incorrect. The fix is to detect if we're on the LHS, and if not, allow read/write references.

Would you be willing to help with this?

shareefj, 2023-07-19:
I'd love to but my knowledge of the Verilator internals is zero and I'm a forced C++ engineer - i.e. only when there's no one capable to do it.  :-)

Any examples of how to detect LHS/RHS?