verilator/verilator #4376: clang compile error: cannot pass object of non-trivial type 'VlAssocArray<IData, VlWide<8>>' (aka 'VlAssocArray<unsigned int, VlWide<8>>') through variadic function; call will abort at runtime

solomatnikov, 2023-07-19:
```
Vtop___024root__DepSet_h84412442__3.cpp:35512:45: error: cannot pass object of non-trivial type 'VlAssocArray<IData, VlWide<8>>' (aka 'VlAssocArray<unsigned int, VlWide<8>>') through variadic function; call will abort at runtime [-Wnon-pod-varargs]
                  vlSymsp->name(),0,vlSelf->top__DOT__testbench__DOT__tl2_mem__DOT__mem);
                                            ^
```
C++ code:
```
        VL_WRITEF("[%0t] Loading %s into %Ntop.load_program\n%1#\n",
                  64,VL_TIME_UNITED_Q(1000),-9,2048,
                  vlSelf->top__DOT__load_program__DOT__testfile.data(),
                  vlSymsp->name(),0,vlSelf->top__DOT__testbench__DOT__tl2_mem__DOT__mem);
```
Verilog:
```
       $display("[%0t] Loading %s into %m\n", $time, testfile, `TESTBENCH_RAM_PATH);
```
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

What 'verilator --version' are you using?  Did you try it with the git master version?
```
v5.012-52-g23fe5c1b9
```
What OS and distribution are you using?
```
Linux loran 4.18.0-500.el8.x86_64 #1 SMP Wed Jun 28 00:07:07 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```
May we assist you in trying to fix this in Verilator yourself?

Maybe

solomatnikov, 2023-07-19:
Tired to this:
```
diff --git a/test_regress/t/t_sys_readmem_assoc.pl b/test_regress/t/t_sys_readmem_assoc.pl
index 493cedf7e..e161a62b9 100755
--- a/test_regress/t/t_sys_readmem_assoc.pl
+++ b/test_regress/t/t_sys_readmem_assoc.pl
@@ -11,6 +11,7 @@ if (!$::Driver) { use FindBin; exec("$FindBin::Bin/bootstrap.pl", @ARGV, $0); di
 scenarios(simulator => 1);
 
 compile(
+    verilator_flags2 => ["--compiler clang"],
     );
 
 execute(
```
but it still uses `g++`.

solomatnikov, 2023-07-21:
> Could you provide the whole verilog code?

Unfortunately, this is impossible

wsnyder, 2023-07-27:
>+ $display("[%0t] assoc_c=%p %m", $time, assoc_c, assoc_c);

I suspect you wanted a %p.

wsnyder, 2023-07-27:
Confirmed works with %p, fails with %m, so really this is bad code that it should flag at verilation time.

solomatnikov, 2023-07-27:
Correct code should be:
```
diff --git a/test_regress/t/t_sys_readmem_assoc.v b/test_regress/t/t_sys_readmem_assoc.v
index 7d2487f31..030d98496 100644
--- a/test_regress/t/t_sys_readmem_assoc.v
+++ b/test_regress/t/t_sys_readmem_assoc.v
@@ -25,7 +25,7 @@ module t(/*AUTOARG*/
       end
       else if (cyc == 2) begin
          $readmemb("t/t_sys_readmem_b.mem", assoc_c);
-         $display("assoc_c=%p", assoc_c);
+         $display("[%0t] assoc_c=%p %m", $time, assoc_c);
          $writememh({`STRINGIFY(`TEST_OBJ_DIR),"/t_sys_writemem_c_b.mem"}, assoc_c);
       end
       else if (cyc == 3) begin
```

closed 2023-11-11 by wsnyder