verilator/verilator #4377: C++ error: redefinition of 'Vtop___024root___eval_initial__TOP__0'

solomatnikov, 2023-07-20:
```
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:14:19: note: previous definition is here
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:8123:19: error: redefinition of 'Vtop___024root___eval_initial__TOP__1'
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__1(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:1390:19: note: previous definition is here
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__1(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:8861:5: error: call to 'Vtop___024root___eval_initial__TOP__0' is ambiguous
    Vtop___024root___eval_initial__TOP__0(vlSelf);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:6360:19: note: candidate function
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:8854:19: note: candidate function
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:10508:5: error: call to 'Vtop___024root___eval_initial__TOP__0' is ambiguous
    Vtop___024root___eval_initial__TOP__0(vlSelf);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:6360:19: note: candidate function
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:8854:19: note: candidate function
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
Vtop___024root__DepSet_heccd7ead__1__Slow.cpp:10501:19: note: candidate function
VL_ATTR_COLD void Vtop___024root___eval_initial__TOP__0(Vtop___024root* vlSelf) {
                  ^
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

No

What 'verilator --version' are you using?  Did you try it with the git master version?
```
v5.012-52-g23fe5c1b9
```
What OS and distribution are you using?
```
Linux loran 4.18.0-500.el8.x86_64 #1 SMP Wed Jun 28 00:07:07 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```
May we assist you in trying to fix this in Verilator yourself?

Maybe

solomatnikov, 2023-07-20:
This error does not happen if the following verilator options are removed:
```
               --output-split 1000000
               --output-split-cfuncs 5000
               --output-split-ctrace 4000
```

wsnyder, 2023-07-25:
Without providing sources you'll need to debug the output code generation.  I'd suggest change the code that makes the "__" numbered function names in  V3Sched.cpp and V3Order.cpp - search for "__" and use e.g. "__a" for the first one "__b" for the second etc. Then verilate and see what function numbers break.  The new breaking name might be e.g Vtop___024root___eval_initial__TOP__a0 - and the __a here will tell you the code that's breaking.  If the problem goes away the various renamings are conflicting and that tells us something too.

wltr, 2023-08-08:
this just started happening in our code base after upgrading to 5.014. so far I could resolve a few of them by refactoring generate statements.

wsnyder, 2023-08-08:
@wltr as described in earlier comments we need a code example that creates it, or debug as described to try to make a pull request to fix it. Thanks.

wsnyder, 2023-08-12:
I suspect is duplicate of #4416 which was just fixed, can you confirm?

solomatnikov, 2023-08-14:
This error doesn't happen at 16008acdf.

It fails later:
```
g++     sim_main.o verilated.o verilated_dpi.o verilated_vcd_c.o verilated_threads.o Vtop__ALL.a    -pthread -lpthread -latomic   -o Vtop
verilated.o: In function `VerilatedContext::addModel(VerilatedModel*)':
verilated.cpp:(.text+0x942f): undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()'
verilated.o: In function `VerilatedContextImp::commandArgVlUint64(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long&, unsigned long, unsigned long)::$_0::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
verilated.cpp:(.text+0xa0e2): undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()'
verilated.o: In function `Verilated::scTimePrecisionError(int, int)':
verilated.cpp:(.text+0xaaf9): undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()'
verilated.o: In function `std::__new_allocator<VerilatedMsg*>::allocate(unsigned long, void const*)':
verilated.cpp:(.text._ZNSt15__new_allocatorIP12VerilatedMsgE8allocateEmPKv[_ZNSt15__new_allocatorIP12VerilatedMsgE8allocateEmPKv]+0x1e): undefined reference to `std::__throw_bad_array_new_length()'
verilated.o: In function `std::__new_allocator<unsigned int>::allocate(unsigned long, void const*)':
verilated.cpp:(.text._ZNSt15__new_allocatorIjE8allocateEmPKv[_ZNSt15__new_allocatorIjE8allocateEmPKv]+0x1e): undefined reference to `std::__throw_bad_array_new_length()'
verilated.o: In function `std::__new_allocator<VerilatedRange>::allocate(unsigned long, void const*)':
verilated.cpp:(.text._ZNSt15__new_allocatorI14VerilatedRangeE8allocateEmPKv[_ZNSt15__new_allocatorI14VerilatedRangeE8allocateEmPKv]+0x1e): undefined reference to `std::__throw_bad_array_new_length()'
verilated.o: In function `std::__new_allocator<std::__detail::_Hash_node_base*>::allocate(unsigned long, void const*)':
verilated.cpp:(.text._ZNSt15__new_allocatorIPNSt8__detail15_Hash_node_baseEE8allocateEmPKv[_ZNSt15__new_allocatorIPNSt8__detail15_Hash_node_baseEE8allocateEmPKv]+0x1e): undefined reference to `std::__throw_bad_array_new_length()'
verilated.o: In function `std::__new_allocator<VerilatedScope const*>::allocate(unsigned long, void const*)':
verilated.cpp:(.text._ZNSt15__new_allocatorIPK14VerilatedScopeE8allocateEmPKv[_ZNSt15__new_allocatorIPK14VerilatedScopeE8allocateEmPKv]+0x1e): undefined reference to `std::__throw_bad_array_new_length()'
verilated.o:verilated.cpp:(.text._ZNSt15__new_allocatorIP8_IO_FILEE8allocateEmPKv[_ZNSt15__new_allocatorIP8_IO_FILEE8allocateEmPKv]+0x1e): more undefined references to `std::__throw_bad_array_new_length()' follow
collect2: error: ld returned 1 exit status
make[1]: *** [Vtop.mk:66: Vtop] Error 1
rm Vtop__ALL.verilator_deplist.tmp
```

wsnyder, 2023-08-15:
@solomatnikov yours are glibc problems unrelated to this, and probably related to something corrupted at your end.

wsnyder, 2023-08-22:
Base issue fixed, as above think other errors are at your end. Feel free to followup if evidence otherwise.

closed 2023-08-22 by wsnyder

solomatnikov, 2023-08-24:
Yes, after linking problem is resolved it is compiled successfully.