verilator/verilator #4379: Multithreading

Shatur, 2023-07-20:
In documentation [Multithreading](https://verilator.org/guide/latest/verilating.html#multithreading) section describes that Verilator model can be generated multithreaded.
Does it mean that threads will be spawned automatically or do I need to create multiple top instances manually?

wsnyder, 2023-07-21:
If you want a single model that is multithreaded verilator does all this for you (this is the common case).  If you want multiple independent models to run in parallel you need to create each yourself (this is rare).

closed 2023-07-21 by wsnyder

Shatur, 2023-07-21:
A few follow-up questions :)

> If you want a single model that is multithreaded verilator does all this for you (this is the common case).

So I just need to pass `--threads` with a number > 1? Because I tried it and it looks like it does not affect the performance in my case...

>  If you want multiple independent models to run in parallel you need to create each yourself (this is rare).

Do I need to specify `--threads` in this case?

wsnyder, 2023-07-21:
>So I just need to pass --threads with a number > 1? Because I tried it and it looks like it does not affect the performance in my case...

Mileage varies.  Often small models (those faster than 1-10 kcycles/sec) don't parallelize, likewise some models with critical paths.  See the documentation on how to use the verilator gantt profiling to see where the bottlenecks are.

>>If you want multiple independent models to run in parallel you need to create each yourself (this is rare).
>Do I need to specify --threads in this case?

The --threads indicates how many threads for each separate model, so if you have 2 models each using 3 threads (6 total threads) pass --threads 3, or nothing for 1 thread.

Shatur, 2023-07-21:
Got it, thank you!

Shatur, 2023-08-08:
@wsnyder one more quick question. Documentation mention that VPI calls should be performed from the main thread. Do I understand correctly that `vpi_register_cb` affects all threads?

Shatur, 2023-08-09:
@wsnyder And the registered callback (on value change, for example) will be called for every module from its thread?

Shatur, 2023-08-10:
Got it, thanks a lot!

Shatur, 2023-08-10:
@wsnyder one more question! Do I need to call `VerilatedVpi::callValueCbs()` from the main thread too?

wsnyder, 2023-08-10:
Am I understand correctly that it's impossible to have multiple independent threads with modules and trigger VPI callbacks for them separately?

That's correct, the IEEE VPI interface is global, there's no such thing as a simulation context.

yulong-lan, 2024-04-27:
Hi @wsnyder. This question here fits my scenario. I'd appreciate it a lot if you could help me confirm if I understand this correctly.  Does it mean that the simulation can basically only run single-threaded when VPI is constantly involved in the simulation?

wsnyder, 2024-04-27:
The simulation can use multiple threads, but any VPI activity will be on only the main thread, so if your VPI uses say 1 unit of compute and your simulation 2 units of compute the runtime will be roughly = `1 + 2/num_threads`..