verilator/verilator #4384: non-blocking assignment in always block acts as blocking?

amiloradovsky, 2023-07-23:
Unfortunately I don't have any of the proprietary simulators handy to verify it, but IIRC there the `addr_wtf` would be delayed by one clock cycle w.r.t. `addr`. While in fact in Verilator it acts as continuous assignment (combinatorial).

```systemverilog
module top;

   localparam int unsigned STEP = 10;
   localparam int unsigned ADDR_WIDTH = 3;

   logic aclock = 0, areset_n = 0;
   logic [ADDR_WIDTH - 1 : 0] addr;
   logic [ADDR_WIDTH - 1 : 0] addr_wtf;

   always #(STEP / 2) aclock = ~aclock;

   initial
     begin
        $dumpfile ("dump.vcd");
        $dumpvars;

        $write ("Resetting... ");
        #(STEP * 10) areset_n = 1;
        $display ("done");
        #(STEP * 10);

        for (int a = 0; a != 1 << ADDR_WIDTH; a++)
          begin
             addr = a[ADDR_WIDTH - 1 : 0];
             @(posedge aclock);
          end

        #(STEP * 1000) $finish;
     end

   always_ff @(posedge aclock)
     addr_wtf <= addr;

endmodule : top

// verilatoR --binary --cc --timing --trace --top top assignment.sv
// obj_dir/Vtop
// gtkwave dump.vcd
```
I'm not sure what I'm doing wrong here^, but it seems like the behavior is incorrect.
```shell
$ verilator --version  # NixOS
Verilator 5.012 2023-06-13 rev UNKNOWN.REV
```

amiloradovsky, 2023-07-25:
```
iverilog -o assignment -g2005-sv assignment.sv
./assignment
gtkwave dump.vcd &
scrot -s
```
[image]
Looks incorrect either^.
```
$ iverilog -V | head -1
Icarus Verilog version 12.0 (stable) ()
```

wsnyder, 2023-07-25:
The addr assignment can race, the @(posedge clk) can be ordered legally before or after the always_ff.  You need to use "addr <= a".  To see what's going on it's better to have a logfile rather than waves as you can see individual events fire

    always @* $display("[%0t] %x %x", $time, addr, addr_wtf);

closed 2023-07-25 by wsnyder

amiloradovsky, 2023-07-25:
Well, `addr <= a[ADDR_WIDTH - 1 : 0];` only triggered the warning
```
%Warning-INITIALDLY: assignment.sv:26:19: Non-blocking assignment '<=' in initial/final block
                                        : ... This will be executed as a blocking assignment '='!
   26 |              addr <= a[ADDR_WIDTH - 1 : 0];
      |                   ^~
                     ... For warning description see https://verilator.org/warn/INITIALDLY?v=5.012
                     ... Use "/* verilator lint_off INITIALDLY */" and lint_on around source to disable this message.
```
And nothing changed in the waveforms :shrug:

amiloradovsky, 2023-07-25:
Meanwhile in Icarus it really made a difference (and there were no warnings).

wsnyder, 2023-07-25:
The warning is telling you it will simulate as "=" given Verilator limitations.  For Verilator you'll need to refactor the code to not be using an initial.

amiloradovsky, 2023-07-25:
OK. Thanks. Guess I gotta use SystemC anyways.

max-kudinov, 2024-04-08:
> The warning is telling you it will simulate as "=" given Verilator limitations. For Verilator you'll need to refactor the code to not be using an initial.

Any plans to support non-blocking assignments in initial? And if not, how can I avoid using inital in my testbenches?

max-kudinov, 2024-04-12:
@wsnyder, as far as I can tell this issue arises even in a simple initial block without a loop, for example:

```SystemVerilog
module veril;
    logic a, b;
    initial begin
        a <= 1;
        b <= 0;
    end
endmodule
```

I get 

```

wsnyder, 2024-04-12:
Ah, as workaround use an event.  We'd welcome a fix for this.

```
module veril;
   logic a, b;
   event e;
   initial begin
      -> e;
   end
   always @(e) begin
      a <= 1;
      b <= 0;
   end
endmodule
```

max-kudinov, 2024-04-12:
Thanks! Could you please give some general info on how to approach this issue and implement a fix? I'm not a C++ programmer, so it is probably out of my skill level, but I'm willing to take a look, and perhaps someone else reading this would find it helpful.

Kreijstal, 2024-06-16:
Ahh, I was going to create a new issue, and after hours of debugging I found this was the culprit... 
You say the solution is to refactor the code, but I am not sure how is the intended refactor, I only know xilinix and questa seem to be able to do this. 

for curiosity this is the code snippet

```sv
task do_transaction();
  integer i;
  // Send request on A channel:
  @(posedge clk_i);
  tl_o.a_valid <= '1;
  @(posedge clk_i);
  i = '0;
  if(tl_o.a_valid) begin
    $display("this ");
  end
  while (!tl_i.a_ready) begin
    i = i + 1;
    if (i == 10) begin
      $display("Warning: device takes > 10 cycles to respond. This is broken on verilator");
    end
    if (i == 1000) begin
      $display("Thing is broken (verilator only).");
      $stop;
    end

    @(posedge clk_i);
  end
  $display("finished with %d",i);
  tl_o.a_valid <= '0;
  tl_o.d_ready <= '1;
  @(posedge clk_i);
  //tl_o <= TlIdle;  // a_valid <= '0;
  while (!tl_i.d_valid) begin
    @(posedge clk_i);
    $display("Waiting for response to be d_valid = 1");
  end
  if (tl_i.d_error) begin
    $display("Warning: response d_error was %p.", tl_i.d_error);
  end
  if (tl_i.d_size != tl_o.a_size) begin
    $display("Warning: response d_size was %p, expected %p.", tl_i.d_size, tl_o.a_size);
  end
  if (tl_i.d_sink != tl_o.a_source) begin
    $display("Warning: response d_sink was %p, expected %p.", tl_i.d_sink, tl_o.a_source);
  end
endtask
```
[complete](https://git.tu-berlin.de/kreijstal/risc-v-lab/-/blob/verilator/src/fv/tlul_test_host.sv?ref_type=heads#L30)

So in here
```
  @(posedge clk_i);
  tl_o.a_valid <= '1;
```
the behaviour of xsim and questa is that tl_o.a_valid changes to 1 only after 1 clock cycle has passed, not instantly, this would be nice in verilator to have, not really sure where to start, but I hope it helps someone

Kreijstal, 2024-06-16:
Not sure if this is the same issue or a related issue?

max-kudinov, 2024-06-17:
> > Thanks! Could you please give some general info on how to approach this issue and implement a fix? I'm not a C++ programmer, so it is probably out of my skill level, but I'm willing to take a look, and perhaps someone else reading this would find it helpful.
> 
> I found something interesting
> 
> https://github.com/verilator/verilator/blob/607c19a67d4a5379c177b1baa034d514de268bf0/src/V3Active.cpp#L365
> 
> maybe check this out?

Seems like it's just a check that gives a warning. 

Perhaps @wsnyder could provide some info on this.

Kreijstal, 2024-06-17:
> Seems like it's just a check that gives a warning.
>
> Perhaps @wsnyder could provide some info on this.

lol yeah that's the check that gives the warning, you have to probably implement a condition so non-blocking is implemented properly, no? I mean maybe not there, but you have a starting point

wsnyder, 2024-06-17:
Yes, that's just to warn there's no implementation if this assumption is violated.  Making it work is the hard part ;)