verilator/verilator #4387: Can't make make_tracing_sc unless VERILATOR_ROOT is set to git install directory

kwmartin, 2023-07-25:
in copy of make_tracing_sc example:
```
> make clean
rm -rf obj_dir logs *.log *.dmp *.vpd coverage.dat core
> export VERILATOR_ROOT=/usr/local
> make

-- Verilator tracing example

-- VERILATE ----------------
/usr/local/bin/verilator -sc --exe -x-assign fast -Wall --trace --assert --coverage -f input.vc top.v sc_main.cpp
%Error: Cannot find verilated_std.sv containing built-in std:: definitions:/usr/local/include/verilated_std.sv
        ... Looked in:
             /usr/local/include/verilated_std.sv
             /usr/local/include/verilated_std.sv.v
             /usr/local/include/verilated_std.sv.sv
             /usr/local/include/verilated_std.sv.vh
             /usr/local/include/verilated_std.sv.svh
             obj_dir//usr/local/include/verilated_std.sv
             obj_dir//usr/local/include/verilated_std.sv.v
             obj_dir//usr/local/include/verilated_std.sv.sv
             obj_dir//usr/local/include/verilated_std.sv.vh
             obj_dir//usr/local/include/verilated_std.sv.svh
%Error: Exiting due to 1 error(s)
make: *** [Makefile:73: run] Error 1
```
If I set VERILATOR_ROOT to verilator git install directory, then example runs fine
```
> make clean
rm -rf obj_dir logs *.log *.dmp *.vpd coverage.dat core
> export VERILATOR_ROOT=/home/Dropbox/programming/Verilog/verilator
> make

-- Verilator tracing example

-- VERILATE ----------------
/home/Dropbox/programming/Verilog/verilator/bin/verilator -sc --exe -x-assign fast -Wall --trace --assert --coverage -f input.vc top.v sc_main.cpp

-- COMPILE -----------------
make -j -C obj_dir -f ../Makefile_obj
make[1]: Entering directory '/home/Dropbox/programming/Verilog/verilator_wrk/examples/make_tracing_sc/obj_dir'
ccache g++  -I.  -MMD -I/home/Dropbox/programming/Verilog/verilator/include -I/home/Dropbox/programming/Verilog/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -I/usr/local/include  -MMD -MP -DVL_DEBUG=1 -Wno-deprecated -Os -fstrict-aliasing -c -o sc_main.o ../sc_main.cpp
ccache g++ -Os  -I.  -MMD -I/home/Dropbox/programming/Verilog/verilator/include -I/home/Dropbox/programming/Verilog/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -I/usr/local/include  -MMD -MP -DVL_DEBUG=1 -Wno-deprecated -c -o verilated.o /home/Dropbox/programming/Verilog/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/home/Dropbox/programming/Verilog/verilator/include -I/home/Dropbox/programming/Verilog/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -I/usr/local/include  -MMD -MP -DVL_DEBUG=1 -Wno-deprecated -c -o verilated_cov.o /home/Dropbox/programming/Verilog/verilator/include/verilated_cov.cpp
ccache g++ -Os  -I.  -MMD -I/home/Dropbox/programming/Verilog/verilator/include -I/home/Dropbox/programming/Verilog/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -I/usr/local/include  -MMD -MP -DVL_DEBUG=1 -Wno-deprecated -c -o verilated_vcd_c.o /home/Dropbox/programming/Verilog/verilator/include/verilated_vcd_c.cpp
ccache g++ -Os  -I.  -MMD -I/home/Dropbox/programming/Verilog/verilator/include -I/home/Dropbox/programming/Verilog/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -I/usr/local/include  -MMD -MP -DVL_DEBUG=1 -Wno-deprecated -c -o verilated_threads.o /home/Dropbox/programming/Verilog/verilator/include/verilated_threads.cpp
/usr/bin/python3 /home/Dropbox/programming/Verilog/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root__DepSet_h84412442__0.cpp Vtop___024root__DepSet_heccd7ead__0.cpp Vtop__Trace__0.cpp Vtop___024root__Slow.cpp Vtop___024root__DepSet_h84412442__0__Slow.cpp Vtop___024root__DepSet_heccd7ead__0__Slow.cpp Vtop__Syms.cpp Vtop__Trace__0__Slow.cpp > Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
ccache g++ -Os -fstrict-aliasing  -I.  -MMD -I/home/Dropbox/programming/Verilog/verilator/include -I/home/Dropbox/programming/Verilog/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -I/usr/local/include  -MMD -MP -DVL_DEBUG=1 -Wno-deprecated -c -o Vtop__ALL.o Vtop__ALL.cpp
Archive ar -rcs Vtop__ALL.a Vtop__ALL.o
g++      -L/usr/local/lib-linux64 sc_main.o verilated.o verilated_cov.o verilated_vcd_c.o verilated_threads.o Vtop__ALL.a    -pthread -lpthread -latomic  -lsystemc -o Vtop
rm Vtop__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/Dropbox/programming/Verilog/verilator_wrk/examples/make_tracing_sc/obj_dir'

-- RUN ---------------------
obj_dir/Vtop +trace

        SystemC 2.3.4-Accellera --- Jul 24 2023 14:38:10
        Copyright (c) 1996-2022 by all Contributors,
        ALL RIGHTS RESERVED
[0] Model running...

Enabling waves into logs/vlt_dump.vcd...
[43000] fastclk is 5 times faster than clk

*-* All Finished *-*
- sub.v:46: Verilog $finish

-- COVERAGE ----------------
/home/Dropbox/programming/Verilog/verilator/bin/verilator_coverage --annotate logs/annotated logs/coverage.dat
Total coverage (5/34) 14.00%
See lines with '%00' in logs/annotated

-- DONE --------------------
To see waveforms, open vlt_dump.vcd in a waveform viewer

> 
```

wsnyder, 2023-07-26:
>%Error: Cannot find verilated_std.sv containing built-in std:: definitions:/usr/local/include/verilated_std.sv

Whoever created your /usr/local/bin/verilator broke it as they set the include path to "definitions:/...." that is the filename includes literally "definitions:" which is incorrect.  Please fix that if you did it yourself, or contact your package maintainer.

Note the way you got around it is hazardous, as it assumes that the verilator version installed corresponds to the include/ files in the git repo checkout, which it may not.  Use the binary built from the git repo also.

closed 2023-07-26 by wsnyder

kwmartin, 2023-07-26:
Thank you, closed!