verilator/verilator #4388: Timing Problem with make_tracing_sc example

kwmartin, 2023-07-26:
Overview: for make_tracing_sc verilator example: adding timing resolution of 1fs runs to completion
However, changing timing resolution to 10fs gives error:
Error Case Changes:
Makefile Added:
```
# Run Verilator in timing mode
VERILATOR_FLAGS += --timing
# Set default time resolution
VERILATOR_FLAGS += --timescale-override /10fs
```
In sc_main.cpp at beginning of sc_main() added:
```
    // define timing resolution as 1ps is way too large
    sc_set_time_resolution(10, SC_FS);
```
And in top.v and sub.v, just before the "module" lines added:
```
`timescale 1ps/10fs
```
running make clean and then make gives:
...
```
-- RUN ---------------------
obj_dir/Vtop +trace

        SystemC 2.3.4-Accellera --- Jul 24 2023 14:38:10
        Copyright (c) 1996-2022 by all Contributors,
        ALL RIGHTS RESERVED
%Error: SystemC's sc_set_time_resolution is 10^-99, which does not match Verilog timeprecision 10^-14. Suggest use 'sc_set_time_resolution(1s)', or Verilator '--timescale-override 1s/1s'
Aborting...
make: *** [Makefile:85: run] Aborted (core dumped)
> 
```
Case that works (I went back and forth quite a few times to verify):
Changes: exactly the same as the case that didn't work, but change all the "10"s to "1"s, that is use:
```
VERILATOR_FLAGS += --timescale-override /1fs
    sc_set_time_resolution(1, SC_FS);
`timescale 1ps/1fs
```
and example runs to completion:
...
```
[0] Model running...

Enabling waves into logs/vlt_dump.vcd...
[43000000] fastclk is 5 times faster than clk

*-* All Finished *-*
- sub.v:47: Verilog $finish

-- COVERAGE ----------------
/home/Dropbox/programming/Verilog/verilator/bin/verilator_coverage --annotate logs/annotated logs/coverage.dat
Total coverage (5/34) 14.00%
See lines with '%00' in logs/annotated

-- DONE --------------------
To see waveforms, open vlt_dump.vcd in a waveform viewer

> 
```
I can live with using a 1fs resolution (1ps is way too large), it just means I have redo my standard digital library; probably not a bit deal, but I still thought it might be of general interest to correct this.
p.s. I'm brand new to SystemC (and I don't really want to pay for the IEEE Reference Manual until I have to), but I don't think I made a mistake.

wsnyder, 2023-07-26:
I believe SystemC requires 1fs time precision.  The error-printed "10^-99" comes directly from SystemC's sc_prec, (indicating SystemC doesn't think it is valid), and I don't see a way Verilator could be mis-implementing that.

wsnyder, 2023-07-26:
P.S. At one point SystemC supported 1fs/1ps etc, that is not e.g. 10fs/100fs, perhaps try that.

closed 2023-07-26 by wsnyder

kwmartin, 2023-07-27:
I also tried 100fs precision and it also gave the 10e-99s resolution error. I am currently just going with `timescale 1ps/1fs which works and requires the minimum changes to my standard digital library which I now have compiling in verilator (had to get rid of deassigns; the library was done a long time ago but is still good for timing estimates). Currently, I am trying to simplify doing an automated cover for a single module at the top of the verilog hierarchy with one clock and yaml specified input and output ports and input signal changes at specified times which covers most of my use cases. Thanks for the comments and I will assume the issue is closed.