verilator/verilator #4391: ../V3Width.cpp:1804: Unlinked when getting bit width of type defined in parameterized class

em2machine, 2023-07-27:
```
XXX@MacBook-Pro-8 tb % verilator --binary --timing -o intf_test intf_test.sv
%Error: Internal Error: intf_test.sv:28:22: ../V3Width.cpp:1804: Unlinked
                                          : ... In instance top
   28 |    xxx_class#(Cfg0)::cmd_tag_t tag, tag_q;
      |                      ^~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```

package pf;
   typedef struct packed {
      int unsigned CcNumTl;
      int unsigned PqSize;
   } cfg_t;
endpackage

virtual class xxx_class #(parameter pf::cfg_t Cfg);
   typedef struct packed {
      logic [$clog2(Cfg.CcNumTl)-1:0] tl_index;
      logic [$clog2(Cfg.PqSize)-1:0] pq_index;
   } cmd_tag_t;
endclass

module mod2  #(parameter p_width=16) (
   output logic [p_width-1:0] q,
   input logic [p_width-1:0] d
);
   assign q = d;
endmodule

module top();
   localparam pf::cfg_t Cfg0 = '{
      CcNumTl:8
      ,PqSize:12
   };

   xxx_class#(Cfg0)::cmd_tag_t tag, tag_q;

   mod2 #($bits(tag)) t0(tag_q, tag);

   initial begin
      #100;
      $finish;
   end
endmodule
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
XXX@MacBook-Pro-8 tb % verilator --version
Verilator 5.013 devel rev v5.012-68-g959387b69
```

What OS and distribution are you using?
```
XXX@MacBook-Pro-8 tb % uname -a
Darwin MacBook-Pro-8.local 22.5.0 Darwin Kernel Version 22.5.0: Mon Apr 24 20:53:19 PDT 2023; root:xnu-8796.121.2~5/RELEASE_ARM64_T6020 arm64
```

May we assist you in trying to fix this in Verilator yourself?

Unsure.  Suspect related to #3441, #2783, #4256

wsnyder, 2023-07-27:
Confirmed.  I agree it is related to class parameter resolution.  Maybe @RRozak can make suggestions.

RRozak, 2023-07-28:
I minimized the example:
```systemverilog
class xxx_class #(parameter int X);
   typedef logic [X:0] cmd_tag_t;
endclass

module mod2 #(parameter p_width=16) ();
endmodule

module top();

   xxx_class#(15)::cmd_tag_t tag;

   mod2 #($bits(tag)) t0();
endmodule
```
The problem is that the type of the variable `tag` is unlinked on the stage in which parameterized modules/classes are instantiated. It wasn't linked before that stage, because the class `xxx_class#(15)` wasn't instantiated yet. Members of parameterized classes are linked in the 2nd pass of V3LinkDot, which is just after the instantiation stage (V3Param). In our case it needs to be linked earlier to properly compute the value of `$bits(tag)`, which is needed to instantiate `t0`.

RRozak, 2023-08-01:
I think that the current way of linking is insufficient for such cases and it has to be changed in order to solve this issue.

One of possible approaches is to perform V3LinkDot, V3Param and V3Width in a loop until no change is done in the AST. Maybe we can optimize it by not processing subtrees in which everything is linked and the width of each node is computed.

Another approach that I think is possible is to create a directed graph in which vertices are AST nodes and an edge from `u` to `v` means that the node `u` requires `v` to be linked first (or have its width computed or be instantiated if it's a parameterized class or a module). Nodes would be processed by traversing this graph by DFS and each node would be visited only once.

Both approaches will require a lot of work, but the first one will require much less than the one with a graph.

wsnyder, 2023-08-03:
LinkDot/Param/Width are already performed multiple times, though not in a true loop.  I hadn't thought of the graph approach and is worth more thought.

These elaboration stages are already very complicated and fragile, so something that maintains decoupling even if more "complicated" like creating a graph (that is easy to visualize/debug) might end up really being less effort/complexity in the end.

solomatnikov, 2023-08-24:
Looks related to #4280