verilator/verilator #4393: Compile large netlist with access to signals in C

gephaistos, 2023-07-28:
Greeting!
I am facing trouble with compilation of this [EdDSA project](https://github.com/dqi/ed25519_fpga) netlist.

wsnyder, 2023-07-28:
I presume you are marking certain signals, not everything?  With everything there is a lot, lot more to do, though even this seems slower than expected.

Are you referring to compilation (gcc/clang) time, or verilation time?

There has been at least one compile speed enhancement since 5.006, please try the last release (or git head), thanks.

gephaistos, 2023-07-28:
Thanks for the quick reply!
Yes, I am marking only part of signals (2-3 at most) in element definitions (e.g. `AND2_X1`), so I can access them in any instance of this cell in netlist. So, yeah, basically there are a lot of tracked signals, though that's what I require.

Sorry that the description wasn't clear enough, I refer to compilation time, verilation takes ~15 minutes, and with metacomments approximately the same amount of time. But with metacomments compilation just hangs... Using gcc 9.4.0.

I'll update tools and try again.

gephaistos, 2023-08-01:
Updated to Verilator version 5.012.
Noticed performance improvements, verilation and compilation runs faster in case without marked signals (~10 and ~25 mins now). Unfortunately, the same problem remains with the metacomments.

Then I decided to experiment with clang and this is somewhat astonishing.
In the first case clang compiled everything in 1 min and it also managed to compile in the second case (1.5 hours). Well, as far as I know clang can be faster than gcc but the difference here is insane. Idk what's going on, is it the difference of the compilers in general or the generated code contains something that gcc struggles to process, and clang handles easily, at least in this particular case netlist (or maybe this is always true and clang should be preferred in complement with Verilator?)

All in all, I guess that the problem is solved somehow and this issue can be closed. Though it feels a bit weird to me that adding signals access causes such a huge compilation time increase, while verilation time is nearly the same, but maybe this is expected.

wsnyder, 2023-08-01:
This indicates a gcc performance bug where likely something goes exponential. Both gcc and clang have had similar bugs in the past, that they have fixes. Could try the latest gcc, and if still is slow report a bug to gcc?

wsnyder, 2023-08-12:
If you do report to gcc please post their ticket number for reference, thanks.

closed 2023-08-12 by wsnyder