verilator/verilator #4400: $monitor prints intermediate values during simulation deltas

amanvm, 2023-07-31:
IEEE Verilog standard says that $monitor displays values at the end of the time step.

> the entire argument list is displayed at the __*end of the time*__ step as if reported by the $display task.

But verilator seems to print intermediate value changes during simulation deltas.

What 'verilator --version' are you using?  Did you try it with the git master version?
```
Verilator 5.012 2023-06-13 rev UNKNOWN.REV
```

No, I did not try the master version.

What OS and distribution are you using?
macOS aarch64

May we assist you in trying to fix this in Verilator yourself?
Not sure what this means, am not in a hurry and can wait for a proper fix.

wallento, 2023-08-04:
Thanks for the report. The question is if you are interested to dig into the Verilator code yourself, which can be a fun adventure. I think the issue, once confirmed, is not overly complex to fix even for a newbie in the codebase.

Can you provide a minimal example that can be used to confirm the issue and that can be used as a regression test. You can find such tests here: https://github.com/verilator/verilator/tree/master/test_regress/t

wsnyder, 2024-07-24:
Thanks for working on this.

closed 2024-07-24 by wsnyder