verilator/verilator #4410: --lib-create option with --timing support

TarasSmarkutskyi, 2023-08-09:
Hi all, I am very interested in the following feature.
We want to develop the behavior models of IP blocks and compile them with the --lib-create option for sharing with users. And it  requires a using delays. 
If I'm not mistaken, now it is not supported.

1) Is it possible to provide this feature?
2) If yes, is it very difficult?
3) If Yes, do you work on it or plan to work?

Thanks.

wsnyder, 2023-08-10:
@kbieganski?

At a minimum, the timing events need to be somehow separated between the library and upper module.  Ideally this would be through adding additional code in the verilog wrapper.  This will require some thought to prototype how that will work, then the effort to implement it, @TarasSmarkutskyi is this something you may be willing to work on?

TarasSmarkutskyi, 2023-08-10:
@wsnyder
Unfortunately, I am not proficient in verilator and I am not C++ programmer. Probably it will be difficult for me. 
But I am very interested to discover it more. 

1) Where I can get more info about how verilator works?  Is there are any info (description, diagrams, etc) about verilating process ?
Or the only way it is verilator debugging? 
2) It would be great to know what are the biggest problems to add support timings in compiled libs. What exactly not enough to do this ? What are the main blocking factors? 

I would be very grateful for a few comments from you.

wsnyder, 2023-08-11:
For more information on internals see docs/internals.rst  https://github.com/verilator/verilator/blob/master/docs/internals.rst

The main blocking factor is the manpower for someone to implement it.  If you/your organization cannot do it yourself, you could consider sponsoring the feature with a paid developer, e.g. one listed in https://www.veripool.org/verilator_commercial_support/

kbieganski, 2023-08-21:
I'm not too familiar with `--lib-create`, so I can't offer a good solution. The reason why delays are disallowed with it is that the C wrapper for a design is not aware of events that happen in between its eval calls.

So when you evaluate your lib, move time forward, and evaluate it again, there may have been a delay inside that should've been handled between those two points.

Compare the [`--timing` eval loop](https://veripool.org/guide/latest/connecting.html#evaluation-loop) to what the wrapper generated with `--lib-create` does.

wsnyder, 2026-10-21:
Closing due to age; there is some support for this in master, if there is a small example which is not supported please file a new issue.

closed 2026-10-21 by wsnyder