verilator/verilator #4412: Extra top module gets added to VCD if verilog top has arguments

TheZoq2, 2023-08-09:
Based on #4249 I was under the impression that by setting the top name argument of my verilated file to "", I would get a VCD file where my verilog top module is the top of the verilog file. However, it seems like this only works if the verilog module has no arguments

With
```verilog
module main(input x);
    reg val = 0;

    initial begin
       $dumpfile("out.vcd");
       $dumpvars(0, main);

       $finish();
    end
endmodule
```
A main.cpp with
```cpp
// Verilated -*- C++ -*-
// DESCRIPTION: main() calling loop, created with Verilator --main

#include "verilated.h"
#include "Vmain.h"

//======================

int main(int argc, char** argv, char**) {
    // Setup context, defaults, and parse command line
    Verilated::debug(0);
    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
    contextp->traceEverOn(true);
    contextp->commandArgs(argc, argv);

    // Construct the Verilated model, from Vtop.h generated from Verilating
    const std::unique_ptr<Vmain> topp{new Vmain{contextp.get(), ""}}; // *NOTE* Setting top to "" here

    // Simulate until $finish
    for(int i = 0; i < 10; i++) {
        // Evaluate model
        topp->eval();
        // Advance time
        contextp->timeInc(1);
    }

    if (!contextp->gotFinish()) {
        VL_DEBUG_IF(VL_PRINTF("+ Exiting without $finish; no events left\n"););
    }

    // Final model cleanup
    topp->final();
    return 0;
}
```
when compiled with
```bash
verilator" "--cc" "--exe" "--build" --top-module main "-Wno-fatal" "--trace" main.v main.cpp
```

The vcd file ends up with a scope called `top` in which my actual top `main` is located
[image]

Removing the input `input x`, thus making the verilog
```verilog
module main(input x);
    reg val = 0;

    initial begin
       $dumpfile("out.vcd");
       $dumpvars(0, main);

       $finish();
    end
endmodule
```

Gives the expected result

I need this for similar reasons as #4249. I want the vcd files coming out of verilator to have a similar structure to my test benches in cocotb in which my top verilog module is the top module.

wsnyder, 2023-08-10:
It does this because it wants a place to put the I/O signals that are in the "top" wrapper:

```
 $scope module top $end
  $var wire  1 # x $end
  $scope module main $end
   $var wire  1 $ val $end
   $var wire  1 # x $end
```

The I/O is also in the level below, so maybe it would be ok to skip them. One complication is this needs to be done at runtime since that is when the "" is known, which might slow down tracing as we don't have currently an "ignore" flag for dumping each variable.  Another alternative which is likely easier is to have a flag not to dump these at verilation time (e.g. --no-trace-top).  You can emulate that behavior now, though with the side effect of losing the I/O completely which I suspect you don't want.  (If we fix verilation the I/O will get split into the wrapped I/O which can have the tracing_off around it and fix this side effect.)

```
module main(
            // verilator tracing_off
            input x
            // verilator tracing_on
            );
    reg val = 0;
    wire x_temp = x;
    initial begin
       $dumpfile("out.vcd");
       $dumpvars(0, main);
       $finish();
    end
endmodule
```

Would you be willing to make a pull request to improve this?

TheZoq2, 2023-08-10:
Sure, I'd be happy to give this a shot!

TheZoq2, 2023-08-10:
I think I might need some guidance for where to get started with this though? Looking at #4235 I think I can figure out where to add options and how tests are usually written here. 

But I'm not sure where I should make the modifications to turn off tracing for those generated signals, or how to write a test to verify that that is actually happening

PR #4422 (2023-08-19): Add --no-trace-top to not trace top signals (#4412)
Fixes #4412

I mostly mimicked #4249 in order to figure out how to add options. Seems to work in my tests, but I don't know how I would write regression tests for it if that is required.

Hopefully my documentation is clear, but I'm not super happy with the wording.
closed 2023-08-19 by wsnyder