verilator/verilator #4415: Strange high impedance behavior

galacticstudios, 2023-08-10:
I have a problem that I've traced to this line of code:

`assign resultOut = passthrough ? resultIn : 16'bz;`

resultIn and resultOut are 16 bit values.

I appreciate that Verilator doesn't fully support z. But when passthrough is 1 and resultIn is 'h0000, resultOut is being assigned 'h3000. I looked at the generated code, and while I see assignments for resultOut0, resultOut1, and so on, there's no code generated for resultOut12 and resultOut13!

Those two values are initialized, but never assigned again.

If I change the code to:

wsnyder, 2023-08-10:
Try compiling your model with VL_DEBUG defined, perhaps you are driving in an incorrect value and this propagates.

Otherwise verilate with --debug, and look at the .tree output files to see how it is optimizing resultOut and where the 0x3000 comes from.  If you can't find out where, we'll need a complete example to assist.

galacticstudios, 2023-08-12:
But is this something that should work?

I want multiple output signals to be connected together. At any one time, only one will be active and the others will be high impedance. Can Verilator simulate that?

wsnyder, 2023-08-12:
>But is this something that should work?  I want multiple output signals to be connected together. At any one time, only one will be active and the others will be high impedance. Can Verilator simulate that?

Yes, that should work, as long as the primary I/O aren't tristates themselves.

galacticstudios, 2023-08-12:
"Primary I/O"? How is any device designated a primary?

In the case, for example, of a CPU's data bus connected to memory chips and peripheral chips, they can all be outputs or inputs at different times. There's no single "primary", especially if the peripherals are able to be bus masters during DMA.

wsnyder, 2023-08-12:
A primary input/output is one that comes out of the top of your model e.g. exposed to the C++ wrapper you connect to, as opposed to an input/output that just connects to a cell instance buried somewhere inside your design.

galacticstudios, 2023-08-13:
Aha!!

That's why! The bogus value is, indeed, on a primary I/O bus!

Thank you.

closed 2023-08-13 by wsnyder