verilator/verilator #4416: Multiple definition error after update to 5.014

Shatur, 2023-08-11:
I have the following compiling error with 5.014:
```
/usr/bin/ld: CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__1__Slow.cpp.o: in function `aes_core___024root___eval_static__TOP__0(aes_core___024roo
t*)':
aes_core___024root__DepSet_h4dea7a1b__1__Slow.cpp:(.text+0x25): multiple definition of `aes_core___024root___eval_static__TOP__0(aes_core___024root*)'; CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.di
r/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp.o:aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp:(.text+0x8f80): first defined here
/usr/bin/ld: CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__2__Slow.cpp.o: in function `aes_core___024root___eval_static__TOP__0(aes_core___024roo
t*)':
aes_core___024root__DepSet_h4dea7a1b__2__Slow.cpp:(.text+0x0): multiple definition of `aes_core___024root___eval_static__TOP__0(aes_core___024root*)'; CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir
/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp.o:aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp:(.text+0x8f80): first defined here
/usr/bin/ld: CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__3__Slow.cpp.o: in function `aes_core___024root___eval_static__TOP__0(aes_core___024roo
t*)':
aes_core___024root__DepSet_h4dea7a1b__3__Slow.cpp:(.text+0x0): multiple definition of `aes_core___024root___eval_static__TOP__0(aes_core___024root*)'; CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir
/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp.o:aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp:(.text+0x8f80): first defined here
/usr/bin/ld: CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__4__Slow.cpp.o: in function `aes_core___024root___eval_static__TOP__0(aes_core___024roo
t*)':
aes_core___024root__DepSet_h4dea7a1b__4__Slow.cpp:(.text+0x0): multiple definition of `aes_core___024root___eval_static__TOP__0(aes_core___024root*)'; CMakeFiles/secworks-aes.dir/CMakeFiles/secworks-aes.dir
/aes_core.dir/aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp.o:aes_core___024root__DepSet_h4dea7a1b__0__Slow.cpp:(.text+0x8f80): first defined here
```

And the same code compiles fine in 5.012.

Looks it caused by code like this:
```verilog
`ifdef ASSIGNS
    reg Z_f /* verilator public_flat_rw */ = 1'b0;
    reg Z_v /* verilator public_flat_rw */ = 1'b0;
    assign Z = Z_f ? Z_v : (A ^ B);
`else
    assign Z = (A ^ B);
`endif
```
It works without `ASSIGNS`, but causes the error above if its defined.

Here is the minimal reproducible example: [example.zip](https://github.com/verilator/verilator/files/12321501/example.zip)
The example using [freepdk](https://github.com/mflowgen/freepdk-45nm/blob/d3be42a72ca6eabff54a4f3be50c16613f69f1b6/stdcells.lib) library converted to verilog and netlist generated from https://github.com/secworks/aes.
It will report the mentioned error if compiled with ` -DASSIGNS` passed to `VERILATOR_ARGS`. And compiles fine without the define.

wsnyder, 2023-08-12:
Confirmed broken with cmake, but this seems to work fine with make, so it's something related to cmake for others to debug

     verilator -j --prefix aes_core --binary --trace-underscore -Wno-UNOPTFLAT --vpi -DASSIGNS -Iverilog/stdcells verilog/aes_core.v main.cpp

wsnyder, 2023-08-12:
@nickelpro might you take a look as you last changed cmake core?

nickelpro, 2023-08-12:
I don't think this is a cmake issue, reverting 837d9f9d771b2fd5975ef260173383c946dfa535 results in the same error (and I don't see how the two could possibly be related).

@wsnyder Your example uses `ar` to assemble an archive , so it ignores the ODR violation.

```
ar -rcs aes_core__ALL.a aes_core.o aes_core___024root__DepSet_h02180e45__0.o aes_core___024root__DepSet_h4dea7a1b__0.o aes_core___024root__DepSet_h4dea7a1b__1.o aes_core___024root__DepSet_h4dea7a1b__2.o aes_core___024root__DepSet_h4dea7a1b__3.o aes_core___024root__DepSet_h4dea7a1b__4.o aes_core___024root__DepSet_h4dea7a1b__5.o aes_core___024root__DepSet_h4dea7a1b__6.o aes_core___024root__DepSet_h4dea7a1b__7.o aes_core___024root__DepSet_h4dea7a1b__8.o aes_core___024root__DepSet_h4dea7a1b__9.o aes_core___024root__DepSet_h4dea7a1b__10.o aes_core___024root__DepSet_h4dea7a1b__11.o aes_core___024root__DepSet_h4dea7a1b__12.o aes_core__main.o aes_core__Dpi.o aes_core___024root__Slow.o aes_core___024root__DepSet_h02180e45__0__Slow.o aes_core___024root__DepSet_h4dea7a1b__0__Slow.o aes_core___024root__DepSet_h4dea7a1b__1__Slow.o aes_core___024root__DepSet_h4dea7a1b__2__Slow.o aes_core___024root__DepSet_h4dea7a1b__3__Slow.o aes_core___024root__DepSet_h4dea7a1b__4__Slow.o aes_core___024root__DepSet_h4dea7a1b__5__Slow.o aes_core___024root__DepSet_h4dea7a1b__6__Slow.o aes_core___024root__DepSet_h4dea7a1b__7__Slow.o aes_core__Syms.o aes_core__Syms__1.o aes_core__Syms__2.o
```

I believe (I'm not a binutils expert) this causes the offending symbol, `aes_core___024root___eval_static__TOP__0`, to simply be overwritten in the symbol table for the archive.

cmake uses `ld` to link all the archives, and so it surfaces the underlying ODR violation.

I think what's actually happening is verilator isn't incrementing the suffix for the `aes_core___024root___eval_static__TOP` functions correctly. I don't enough about verilator internals to go farther than that

wsnyder, 2023-08-12:
>cmake uses ld to link all the archives, and so it surfaces the underlying ODR violation.

Great point, I'll add an assertion to check for that too.

closed 2023-08-12 by wsnyder