verilator/verilator #4417: Streamline implementation of SystemC mode

gezalore, 2023-08-11:
I find myself coming into conflict with the SystemC mode often while trying to make changes to the model structure. I would like to know if you would accept a change which generates the SystemC module as an add-on wrapper around the C module, instead of the current way of emitting something different. In other words: `verilator --sc` would do the same as `verilator --cc` + generate an additional SC module which wraps an instance of the `--cc` module.

In practice this might require copying the IO signals and structuring some of the code in some slightly different way so they can be integrated. This should have no effect on C mode, and minimal effect on SC performance on large designs. This seems like a good trade-off to keep maintenance and model architecture cleaner. My hope would be that almost nothing except the wrapper emitter should need to know about generating for SC. Maybe there are some technical reasons this is infeasible?

wsnyder, 2023-08-11:
There already is a wrapper, that's "top" which is what maps the SC signals to the internal signals.  That's really the only reason it (originally) exists.  So I'm a bit confused what's proposed - a new sctop then top then the model?  We should not change the SC hierarchy observed by users, I think that may be difficult.

gezalore, 2023-08-11:
Currently there are 2 things:
1. The user visible subclass of `VerilatedModel`, which provides the interface
2. The actual implementation `VerilatedSyms`, containing all static instances, including the instance of "top" (which is SV `$root`)

In SC mode, the user class (point 1) is also an `sc_module` (multiple inheritance from both `sc_module` and `VerilatedModel`), and there are still SC signals inside the `VerilatedSyms`, inside the "top" module, which contains the IO signals.

The suggestion was to have the `sc_module` instantiate the --cc `VerilatedModel` instead (currently it instantiates `VerilatedSyms` directly), so `VerilatedSyms` and the top level module need not know about SC, and need not have SC signals for example.

At that point the only thing that internally cares about SC properties and behaviours is the emit pass that writes out the `sc_module`, which hooks it up properly to the --cc `VerilatedModel`.

This does mean that the emitted `sc_module` would no longer be a subclasss of `VerilatedModel`. `VerilatedModel` is a relatively new thing though, added by myself mid last year. The rest of the interface can be kept compatible.

wsnyder, 2023-08-12:
I don't see how to do what you propose without having a relatively slow interface layer similar to how we do --hierarchical, otherwise it would need the scheduling information which would probably intrude more than the current code does.  Perhaps you could make a hand-crafted example wrapper showing otherwise?

>This does mean that the emitted sc_module would no longer be a subclasss of VerilatedModel. VerilatedModel is a relatively new thing though, added by myself mid last year. The rest of the interface can be kept compatible.

I agree I don't think that matters.

gezalore, 2023-08-18:
So the sc_module's `::eval()` would be something like this:
```c++
void Vtop::eval() {
    model->symsp->TOP->input /* regular signal */ = this->input /* sc_in */;
    model->eval();
    this->output /* sc_out */ = model->symsp->TOP->output /* regular signal*/;
}
```

Note that these assignments already exist anyway, as they are necessary to copy the sc_in/sc_out contents to/from internal signals. Currently they are just part of the generated code ('ico' for inputs, 'act' or 'nba' for outputs).

For the trivial example:
```verilog
   module top(
     input wire clk,
     input wire i,
     output reg o
   );

     always @(clk) o = i;

   endmodule
```

We currently generate:
```c++
   VL_INLINE_OPT void Vtop___024root___ico_sequent__TOP__0(Vtop___024root* vlSelf) {
       if (false && vlSelf) {}  // Prevent unused
       Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
       VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___ico_sequent__TOP__0\n"); );
       // Body
       VL_ASSIGN_ISI(1,vlSelf->__Vcellinp__top__clk, vlSelf->clk);
       VL_ASSIGN_ISI(1,vlSelf->__Vcellinp__top__i, vlSelf->i);
   }

   VL_INLINE_OPT void Vtop___024root___nba_sequent__TOP__0(Vtop___024root* vlSelf) {
       if (false && vlSelf) {}  // Prevent unused
       Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
       VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___nba_sequent__TOP__0\n"); );
       // Body
       vlSelf->__Vcellout__top__o = vlSelf->__Vcellinp__top__i;
       VL_ASSIGN_SII(1,vlSelf->o, vlSelf->__Vcellout__top__o);
   }
```

wsnyder, 2023-08-18:
>this->output /* sc_out */ = model->symsp->TOP->output /* regular signal*/;

In SystemC the output assignment is overloaded to schedule a change to all SystemC blocks that read that signal, so doing this always (vs just when Verilator scheduling needs it) would be extremely compute intensive.

closed 2023-08-18 by gezalore

gezalore, 2023-08-18:
Fwiw I checked and the updates are only scheduled if the value is actually changed, so in the grand scheme of things this might not be as extreme as suggested, but I dont have a benchmark to try this out on.

gezalore, 2025-08-28:
Just recording some information. See also #6329.

As for the efficiency concerns, realize that in all well-formed `--sc` output we generate, there exists an assignment for each top level output port of the form `scOutputPort = internalVariable` where the internal variable is either the output of the top level module. This is effectively created in `V3LinkLevel::wrapTopCell` by making an `AstPin` for the output on the root module (which instantiates the top level module).

This means there already is a combinational "copy" involved for all SC output ports. The difference is that that copy might only be executed when one of the event domains (clocks) triggers that combinational update.

We could either:
1. add a shadow copy of the `internalVariable` in the SC wrapper proposed in this issue that only writes the `scOutputPort` when the value changed, however, SC already does this when calling .write() on a signal and won't post an update unless env `SC_SIGNAL_WRITE_CHECK` is set to `CONFLICT` (see sc_signal_t::write, easiest is to step there through GDB in t_sc_vl_assign_sbw)
2. Add an extra bit for each output, that is reset to 0 on entering eval, and set to 1 in the continuous assignment assigning the top level output. That way it's easy to see when an output signal *might have* changed.