verilator/verilator #4418: Vfork duplicate functions

wsnyder, 2023-08-12:
Debugging #4416 lead me to add an assertion to V3Broken to look for duplicate functions.

In master, turn this on by removing this if statement in V3Broken.cpp:

        if (false) { // bug4118

Then running regressions:

    t/t_timing_sched.v:10:19: ../V3Broken.cpp:288: Duplicate cfunc name: '__Vfork_1__0'

As #4416 noticed this may go unnoticed unless using cmake (unless the above assertion is on), so this is a real potential problem not just an assertion issue.

@RRozak I think you were last in there for #4295, mind taking a look when get a chance?

RRozak, 2023-08-17:
I reproduced it on a small example:
```systemverilog
module t;
   logic clk1 = 0;
   assign #3 clk1 = ~clk1;
endmodule
```
It is not related to the naming of forks. The problem is with moving statements to separate functions.
The fork is created by Verilator to handle the delayed assignment.
`begin` blocks representing bodies of forks are moved to separate functions here: https://github.com/verilator/verilator/blob/cbdee5a80499aaae049e955b7b45d5d8abc5b8ef/src/V3SchedTiming.cpp#L364-L365
In this example the fork is visited two times, because its copies are present in two functions: `_stl_comb__TOP__0` and `_act_sequent__TOP__0`. I think that the root problem is that this fork is inside these 2 functions.

RRozak, 2023-08-21:
I unassigned myself, because I can't continue the work on it now.
What is left is to check if the fork should really be in both of these functions. If it is correct, we can remove the name conflict by adding a unique number to its name as in other places.

closed 2023-08-30 by wsnyder