verilator/verilator #4423: I'm having trouble linking the xxx.a static library generated by verilator in my systemc project

Alan-19950616, 2023-08-16:
## my CMakeLits.txt like this:
[image]

## the error message like this:
[image]

Alan-19950616, 2023-08-16:
## When I added "verilated.cpp" to my project, there is only one error left
[image]
[image]

wsnyder, 2023-08-16:
Please post text not images, OCR them if needed.

This seems an internal problem with the systemc libraries and not something Verilator has control over, perhaps your library is broken, see if SystemC works outside verilator.

Alan-19950616, 2023-08-16:
I'm sorry about the picture, my systemc's library should be complete and all the examples are passed.
```
Test project /home/nuclei/FastModel/systemc-2.3.4/build/examples
      Start  1: examples/sysc/2.1/dpipe/dpipe
 1/33 Test  #1: examples/sysc/2.1/dpipe/dpipe ................................   Passed    0.10 sec
      Start  2: examples/sysc/2.1/forkjoin/forkjoin
 2/33 Test  #2: examples/sysc/2.1/forkjoin/forkjoin ..........................   Passed    0.02 sec
      Start  3: examples/sysc/2.1/reset_signal_is/reset_signal_is
 3/33 Test  #3: examples/sysc/2.1/reset_signal_is/reset_signal_is ............   Passed    0.05 sec
      Start  4: examples/sysc/2.1/sc_export/sc_export
 4/33 Test  #4: examples/sysc/2.1/sc_export/sc_export ........................   Passed    0.03 sec
      Start  5: examples/sysc/2.1/sc_report/sc_report
 5/33 Test  #5: examples/sysc/2.1/sc_report/sc_report ........................   Passed    0.05 sec
      Start  6: examples/sysc/2.1/scx_barrier/scx_barrier
 6/33 Test  #6: examples/sysc/2.1/scx_barrier/scx_barrier ....................   Passed    0.05 sec
      Start  7: examples/sysc/2.1/scx_mutex_w_policy/scx_mutex_w_policy
 7/33 Test  #7: examples/sysc/2.1/scx_mutex_w_policy/scx_mutex_w_policy ......   Passed    0.05 sec
      Start  8: examples/sysc/2.1/specialized_signals/specialized_signals
 8/33 Test  #8: examples/sysc/2.1/specialized_signals/specialized_signals ....   Passed    0.03 sec
      Start  9: examples/sysc/2.3/sc_rvd/sc_rvd
 9/33 Test  #9: examples/sysc/2.3/sc_rvd/sc_rvd ..............................   Passed    0.05 sec
      Start 10: examples/sysc/2.3/sc_ttd/sc_ttd
10/33 Test #10: examples/sysc/2.3/sc_ttd/sc_ttd ..............................   Passed    0.06 sec
      Start 11: examples/sysc/2.3/simple_async/simple_async
11/33 Test #11: examples/sysc/2.3/simple_async/simple_async ..................   Passed    1.09 sec
      Start 12: examples/sysc/fft/fft_flpt/fft_flpt
12/33 Test #12: examples/sysc/fft/fft_flpt/fft_flpt ..........................   Passed    0.04 sec
      Start 13: examples/sysc/fft/fft_fxpt/fft_fxpt
13/33 Test #13: examples/sysc/fft/fft_fxpt/fft_fxpt ..........................   Passed    0.08 sec
      Start 14: examples/sysc/fir/fir
14/33 Test #14: examples/sysc/fir/fir ........................................   Passed    0.06 sec
      Start 15: examples/sysc/fir/fir_rtl
15/33 Test #15: examples/sysc/fir/fir_rtl ....................................   Passed    0.06 sec
      Start 16: examples/sysc/pipe/pipe
16/33 Test #16: examples/sysc/pipe/pipe ......................................   Passed    0.04 sec
      Start 17: examples/sysc/pkt_switch/pkt_switch
17/33 Test #17: examples/sysc/pkt_switch/pkt_switch ..........................   Passed    0.05 sec
      Start 18: examples/sysc/risc_cpu/risc_cpu
18/33 Test #18: examples/sysc/risc_cpu/risc_cpu ..............................   Passed    0.04 sec
      Start 19: examples/sysc/rsa/rsa
19/33 Test #19: examples/sysc/rsa/rsa ........................................   Passed    0.09 sec
      Start 20: examples/sysc/simple_bus/simple_bus
20/33 Test #20: examples/sysc/simple_bus/simple_bus ..........................   Passed    0.03 sec
      Start 21: examples/sysc/simple_fifo/simple_fifo
21/33 Test #21: examples/sysc/simple_fifo/simple_fifo ........................   Passed    0.06 sec
      Start 22: examples/sysc/simple_perf/simple_perf
22/33 Test #22: examples/sysc/simple_perf/simple_perf ........................   Passed    0.06 sec
      Start 23: examples/tlm/at_1_phase/at_1_phase
23/33 Test #23: examples/tlm/at_1_phase/at_1_phase ...........................   Passed    0.08 sec
      Start 24: examples/tlm/at_2_phase/at_2_phase
24/33 Test #24: examples/tlm/at_2_phase/at_2_phase ...........................   Passed    0.11 sec
      Start 25: examples/tlm/at_4_phase/at_4_phase
25/33 Test #25: examples/tlm/at_4_phase/at_4_phase ...........................   Passed    0.10 sec
      Start 26: examples/tlm/at_extension_optional/at_extension_optional
26/33 Test #26: examples/tlm/at_extension_optional/at_extension_optional .....   Passed    0.06 sec
      Start 27: examples/tlm/at_mixed_targets/at_mixed_targets
27/33 Test #27: examples/tlm/at_mixed_targets/at_mixed_targets ...............   Passed    0.05 sec
      Start 28: examples/tlm/at_ooo/at_ooo
28/33 Test #28: examples/tlm/at_ooo/at_ooo ...................................   Passed    0.07 sec
      Start 29: examples/tlm/lt/lt
29/33 Test #29: examples/tlm/lt/lt ...........................................   Passed    0.08 sec
      Start 30: examples/tlm/lt_dmi/lt_dmi
30/33 Test #30: examples/tlm/lt_dmi/lt_dmi ...................................   Passed    0.07 sec
      Start 31: examples/tlm/lt_extension_mandatory/lt_extension_mandatory
31/33 Test #31: examples/tlm/lt_extension_mandatory/lt_extension_mandatory ...   Passed    0.03 sec
      Start 32: examples/tlm/lt_mixed_endian/lt_mixed_endian
32/33 Test #32: examples/tlm/lt_mixed_endian/lt_mixed_endian .................   Passed    0.06 sec
      Start 33: examples/tlm/lt_temporal_decouple/lt_temporal_decouple
33/33 Test #33: examples/tlm/lt_temporal_decouple/lt_temporal_decouple .......   Passed    0.04 sec

100% tests passed, 0 tests failed out of 33

Total Test time (real) =   3.09 sec

[100%] Built target check
```

Alan-19950616, 2023-08-16:
My project works fine until I link the static libraries generated by verilator, which proves the integrity of the systemc library.

```
nuclei@nuclei-v:~/FastModel/RISC-V-TLM$ cd build/
nuclei@nuclei-v:~/FastModel/RISC-V-TLM/build$ cmake -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0")  
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen missing components: dot
Doxygen build started
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nuclei/FastModel/RISC-V-TLM/build
nuclei@nuclei-v:~/FastModel/RISC-V-TLM/build$ make
[  5%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/A_extension.cpp.o
[ 10%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/BASE_ISA.cpp.o
[ 15%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/BusCtrl.cpp.o
[ 21%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/CPU.cpp.o
[ 26%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/C_extension.cpp.o
[ 31%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Debug.cpp.o
[ 36%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Instruction.cpp.o
[ 42%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/M_extension.cpp.o
[ 47%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Memory.cpp.o
[ 52%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/MemoryInterface.cpp.o
[ 57%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Performance.cpp.o
[ 63%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/RV32.cpp.o
[ 68%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/RV64.cpp.o
[ 73%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Registers.cpp.o
[ 78%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Simulator.cpp.o
[ 84%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Timer.cpp.o
[ 89%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/Trace.cpp.o
[ 94%] Building CXX object CMakeFiles/RISCV_TLM.dir/src/extension_base.cpp.o
[100%] Linking CXX executable RISCV_TLM
[100%] Built target RISCV_TLM
nuclei@nuclei-v:~/FastModel/RISC-V-TLM/build$ ./RISCV_TLM ../tests/C/dhrystone/dhrystone.hex
        SystemC 2.3.4-Accellera --- Aug 14 2023 14:38:55
        Copyright (c) 1996-2022 by all Contributors,
        ALL RIGHTS RESERVED
file: ../tests/C/dhrystone/dhrystone.hex
02 extended address 0x10000
03 PC set to 0x100a6
Created CPURV32 CPU
Get DMI_PTR 
Get DMI_PTR 
Info: /OSCI/SystemC: Simulation stopped by user.
Info: /OSCI/SystemC: Simulation stopped by user.
Total elapsed time: 1.67268s
Total elapsed time: 1.67335s
Simulated 6470566 instr/sec
Simulated 6473157 instr/sec
Press Enter to finish
Press Enter to finish
nuclei@nuclei-v:~/FastModel/RISC-V-TLM/build$ 
```

wsnyder, 2023-08-16:
Check if the systemc headers verilator is using to make the library are different from the headers used for the systemc library.

Alan-19950616, 2023-08-16:
The systemc of the verilator compiler library is the same version(v2.3.4) as the systemc of the test environment.

closed 2023-08-17 by Alan-19950616