verilator/verilator #4424: --x-initial-edge triggers unwanted edge on first model eval

Dolu1990, 2023-08-16:
Hi,

--x-initial-edge triggers unwanted edge on first model eval.
Happen in verilator upstream Verilator 5.015 devel rev v5.014-11-g6c8045726
Doesn't happen in Verilator 4.218 2022-01-17 rev v4.218

Toplevel : 
```verilog
module top
  (input logic clk);

   always_ff @ (posedge clk) begin
     assert(0); //Used as a way to know if posedge clk was triggered
   end
endmodule
```

Test
```c++ 
    Vtop* top = new Vtop; 
    //This eval will trigger always_ff @ (posedge clk) when --x-initial-edge is used (it shouldn't' ?)
    top->eval(); 
    top->final();
```

You can find in example-systemverilog.tar.gz a way to reproduce : 
`make clean run`

Tested on Debian.

The main issue i have with this upstream behaviour, is that clocked assertion are getting triggered at the first eval, without necessarly having a change to reset the system first.

Thanks ^^

wsnyder, 2023-08-19:
--x-initial-edge is intended to cause the posedge to evaluate.  Likely this is not needed any longer with v5 scheduling, does this case run ok without it?  If so, can you see if you can eliminate using it?

Dolu1990, 2023-08-19:
Hi @wsnyder,

> --x-initial-edge is intended to cause the posedge to evaluate. Likely this is not needed any longer with v5 scheduling, does this case run ok without it? 

Yes, without --x-initial-edge it does it right.

> If so, can you see if you can eliminate using it?

The issue is that the absence of  --x-initial-edge create new issues in other cases. 

For instance, if you have the following hardware : 
- async reset -> reset controller which async assert / sync deassert its output -> flip flop with async reset behaviour
What happen is that if the reset controller happen by chance to be initialized in a "asserted output" state, then the flip flop using it as a reset source will not be reseted in verilator sim.

Here is an example how to recreate that issue : 
[example-systemverilog.tar.gz](https://github.com/verilator/verilator/files/12385779/example-systemverilog.tar.gz)
(that design is async assert / sync deasert buffer, connected to a flipflop which should reset with the value 0x42

Expected (what i get with --x-initial-edge) : 
```
-- RUN ---------------------
obj_dir/Vtop
flipflop value after reset 42
<3 Success <3

-- DONE --------------------
 ```

Got (what i get without --x-initial-edge) : 

```
-- RUN ---------------------
obj_dir/Vtop
flipflop value reset 2
!!! Bad value flipflop reset :( !!!
make: *** [Makefile:91: run] Error 1
```

So
- either i use --x-initial-edge, and get "bad" posedge evaluation on the first eval, 
- either i do not use --x-initial-edge, and i get parts of the design which may not trigger on an asserted reset

I can't find a solution to make both cases happy

Dolu1990, 2023-12-18:
>[status: asked reporter](https://github.com/verilator/verilator/labels/status%3A%20asked%20reporter)

Can be removed

wsnyder, 2023-12-18:
I would suggest making your negedge reset pulse after time zero, as I don't expect we will get to resolving this soon.

Dolu1990, 2023-12-22:
> I would suggest making your negedge reset pulse after time zero, as I don't expect we will get to resolving this soon.

I think this would work for flip flop which directly use the toplevel input reset, but if you have some reset controller in the way, it is tricky (assuming --x-initial-edge) : 
- The reset controller internal buffer may  already start in a asserted state
- toplevel input reset negedge after time zero will not make those internal  buffer negedge aswell (unless you wait long enough for all the internal reset buffer to complet their full reset sequance)

wsnyder, 2023-12-22:
True, but what you describe is seeming to describe the way hardware really works, that is flops power up randomly, which is what Verilator models.  Obviously there's no "power on edge" or "X" in HW.  (Ignoring some analog power on effects which are indeterminate, which isn't what X really is made to model.)

Dolu1990, 2023-12-22:
Hmm, that's one thing : 
- Are asyncronous reset forcing the flip flop value "constantly" when asserted, or only on their edge (like described in verilog)

In other words, "is using edge to specify async reset in verilog true to hardware, or is it an artefact of the verilog event driven paradigm"

Looking now at some online schematic of flipflop, i would say it is more an verilog artefact, but i don't know so much about it.

wsnyder, 2025-09-21:
Believed fixed in master with recent timing changes; generally --x-initial is now not required. The provided tarball example from here passes.  If there's still problems please file a new issue, thanks.