verilator/verilator #4426: support dot notation access to types defined inside interfaces

jeras, 2023-08-18:
The SystemVerilog 2017 LRM section _25.10 Access to interface objects_ explicitly defines access to types defined inside interfaces.

When trying to do this in Verilator I get:
```
%Error-UNSUPPORTED: ../../submodules/tcb/hdl/rtl/lib/tcb_lib_demultiplexer.sv:55:3: Unsupported: SystemVerilog 2005 typedef in this context
   55 |   typedef sub.rsp_t rsp_t;
```
But this is a generic error, not related to the described feature. I tried to corrupt the interface name and got the same error.

I can create a testcase for this feature, If you have an existing testcase for updating or copy-reference in mind please tell, otherwise I will try to find out myself in the following days.

jeras, 2023-08-18:
I had a look at existing `test_regress` tests, and `t_interface_typedef.v` seems to cover my feature request.
So if this is a duplicate just close it.

wsnyder, 2023-08-19:
Yes, that's the appropriate test.  I think this duplicates #3441

closed 2023-08-19 by wsnyder

em2machine, 2025-09-18:
@wsnyder sorry for the necro bump.  This was closed as 'completed', was this functionality implemented?   I guess going back and re-reading these issues it appeared to me that it was..  Am I missing something obvious?
Thanks,
-e-

```
%Error-UNSUPPORTED: common/rtl/axi/soc/axi_lite_regs_wrap.sv:17:4: Unsupported: SystemVerilog 2005 typedef in this context
   17 |    typedef axil_tgt_if.addr_t addr_t;
      |    ^~~~~~~
```

```
module axi_lite_regs_wrap #(
   parameter axi4::axi4l_cfg_t AxiCfg=0
)(
   `CNR_PDEF
   ,axi4l_if.target axil_tgt_if
);
   localparam type uint32_t = int unsigned;
   typedef axil_tgt_if.addr_t addr_t;
```

em2machine, 2025-09-18:
Sorry, this issue appears to be about the test, not the functionality.