verilator/verilator #4427: Generate parameter flagged as unused by lint mode.

bpringlemeir, 2023-08-18:
$ **verilator --version**
Verilator 5.012 2023-06-13 rev v5.012-34-g4f13c4d1b
$ **cat dut.v**
```
`timescale 1ns / 10ps
`default_nettype none

module dut
  #( parameter SAMPLE_SIZE = 24 )
   (
    input [4:0] bit_counter,
    output      data_done
    );
   generate
      case (SAMPLE_SIZE)
        16: begin : BIT_COUNTER
           localparam MAX  = 4'd15;
        end
        24: begin : BIT_COUNTER
           localparam MAX  = 5'd23;
        end
        32: begin : BIT_COUNTER
           localparam MAX  = 5'd31;
        end
      endcase // case (SAMPLE_SIZE)
   endgenerate

   assign data_done = (bit_counter >= BIT_COUNTER.MAX);

endmodule // dut
```
$ **verilator -Wall -Wfuture-UNUSEDPARAM -Wno-PINCONNECTEMPTY +1364-2005ext+v -y . --lint-only  dut.v**
%Warning-UNUSEDPARAM: dut.v:16:23: Parameter is not used: 'MAX'
                                 : ... In instance dut
   16 |            localparam MAX  = 5'd23;
      |                       ^~~
                      ... For warning description see https://verilator.org/warn/UNUSEDPARAM?v=5.012
                      ... Use "/* verilator lint_off UNUSEDPARAM */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)

Also updated to 5.014 and a similar issue occurs.  

$ **verilator --version**
Verilator 5.014 2023-08-06 rev v5.014-35-ge6b0bdd4d

I have reduced the test case significantly.  The issue is that in 'verilog' to find a maximum for a parameterized module, I am using a generate case for localparam.  I think this is the trigger for the issue (but it could be even simpler).

bpringlemeir, 2023-08-18:
Quartus does not accept this without the generate stanzas.  However, both Icarus verilog and verilator do.  I minimized to this,

```
`timescale 1ns / 10ps
`default_nettype none

module dut
  #( parameter SAMPLE_SIZE = 24 )
   (
    input [4:0] bit_counter,
    output      data_done
    );
      case (SAMPLE_SIZE)
        32: begin : BIT_COUNTER
           localparam MAX  = 5'd31;
        end
        24: begin : BIT_COUNTER
           localparam MAX  = 5'd23;
        end
      endcase // case (SAMPLE_SIZE)

   assign data_done = (bit_counter >= BIT_COUNTER.MAX);

endmodule
```
This does not produce a warning.  However, if you put case 24, before 32, then there is a warning.  I hope that is helpful.

bpringlemeir, 2023-08-18:
I see, I may have misused labels to cause this.

```
module dut
  #( parameter SAMPLE_SIZE = 24 )
   (
    input [4:0] bit_counter,
    output      data_done
    );
      case (SAMPLE_SIZE)
        24: begin : BIT_COUNTER24
           localparam MAX  = 5'd23;
        end
        32: begin : BIT_COUNTER32
           localparam MAX  = 5'd31;
        end
      endcase // case (SAMPLE_SIZE)

   assign data_done = (bit_counter >= BIT_COUNTER24.MAX);
endmodule
```

This does not produce a warning.

bpringlemeir, 2023-08-21:
I think that the original is legitimate.  IEEE 1800-2012 section 27.5 Conditional generate constructs: "Because at most one of the alternative generate blocks is instantiated, it is permissible for there to be more than one block with the same name within a single conditional generate construct."  The simplified versions without the generate are not.  It may be worked around by disabling with "lint_off UNUSEDPARAM".

wsnyder, 2023-08-21:
Agreed this is legal.  I propose to add a map of what parameter names are used and suppress when there's a name match like this. There's some small chance of a false negative, but this is rare enough it seems reasonable.

bpringlemeir, 2023-08-24:
I think it might be sufficient to document the case in the help text.  Ie, add a  /* verilator lint_off UNUSEDPARAM */ when defining localparams in a generate block with identical labels at https://verilator.org/warn/UNUSEDPARAM?v=5.014  Just to encourage that this is a false positive.  Of course, if you have some other use cases for the map of parameters, then maybe that makes sense.  I wouldn't introduce it just for this issue?

closed 2023-08-25 by wsnyder