verilator/verilator #4433: Passing by reference to task not behaving correctly

njgerber, 2023-08-22:
Compiling the following module (called `my_test.sv`) with `verilator --binary --trace --assert -Wno-PINCONNECTEMPTY my_test.sv` and running it with `./obj_dir/Vmy_test` yields at the end `Should not get here, the task blocks indefinately` while it should be `Task exited`. The reason for it is that the passing by reference to the task does not work and it is just passed by value.

The received output is
```
Starting task
0
1
Should not get here, the task blocks indefinately
```

while it should be

```
Starting task
0
1
2
Task exited
```

```
`default_nettype none

module my_test();
    var logic   rst_bi;
    var logic   clk_i;

    initial begin
        wait (rst_bi == 1'b1);
        while (rst_bi == 1'b1) begin
            clk_i = ~clk_i;
            #10;
        end
    end

    task automatic test_task(ref logic clk);
        $display("0");
        wait (clk == 1'b0);
        $display("1");
        wait (clk == 1'b1);
        $display("2");
    endtask

    initial begin
        $dumpfile("test.vcd");
        $dumpvars;
        rst_bi = 1'b0;
        clk_i = 1'b0;
        #10;
        rst_bi = 1'b1;
        #10;
        wait (clk_i == 1'b0);
        $display("Starting task");
        test_task(clk_i);
        $display("Task exited");
        #100;
        $finish;
    end

    initial begin
        #1000;
        $display("Should not get here, the task blocks indefinately");
        $finish;
    end
endmodule

`default_nettype wire

```

wsnyder, 2023-08-24:
Indeed, it should stay as a C++ reference presumably. Most of the handling code for this predates being able to understand references.

Perhaps you'd be willing to contribute a pull request to fix this (specifically pass the references into the C++ types in the member function declaration)?

njgerber, 2023-08-28:
Yes, I could try to provide a fix for this issue.  Could you please provide me a good starting point as I am not familiar with this project?

wsnyder, 2023-08-28:
V3Task currently checks for isInoutIsh() when making a connection to a task. I think for portp->direction().isRefOrConstRef() it should instead make a temporary variable that is a reference, and set that reference to the object passed.

I think you'll need at least three test cases that use ref to hit the main code paths, one for a normal function that gets inlined (default), and one for a normal function that doesn't inline (`/*verilator no_inline_task*/`), and one for a method.  (The code path followed for last two should in theory be the same.)

wsnyder, 2023-08-28:
And for general information on internals, see docs/internals.rst.  I'd suggest starting by writing a test_regress test for this (with the three cases mentioned) and making sure it passes on another simulator and fails on Verilator due to this issue.

wsnyder, 2023-08-30:
Found an earlier identical issue - please discuss fixes there.  #3385

closed 2023-08-30 by wsnyder