verilator/verilator #4434: I want to add a verilator-generated uart model to systemc project, but there's a problem.

Alan-19950616, 2023-08-23:
I'm not sure if it's a systemc problem or a verilator problem.

Error message

```
Warning: (W116) channel doesn't have a default event
In file: /home/nuclei/FastModel/systemc-2.3.4/src/sysc/communication/sc_interface.cpp:54
pure virtual method called
terminate called without an active exception
```

For details, click on this link
https://github.com/mariusmm/RISC-V-TLM/discussions/24

wsnyder, 2023-08-24:
That seems a SystemC error unrelated to Verilator. There's a lot of complicated stuff in the example you provided, can you try it with a simple sc_main like in the Verilator examples?

wsnyder, 2023-08-30:
Assuming it's something SC related so closing, feel free to followup if determine else wise.

closed 2023-08-30 by wsnyder