verilator/verilator #4442: DPI-C: How to wait for a signal to change?

abarajithan11, 2023-08-26:
I have an SV testbench, where I import a C function and call in an initial block. 

```
// top_tb.sv

module_top_tb;
  bit temp = 0;
  import "DPI-C" function void run_c (inout bit temp);
  initial #1000ns temp=1;
  initial run_c(temp);
endmodule
```

```
// run_c.c

#include <svdpi.h>
extern void run_c (svBit* temp) {
  // do something when temp=0;
  while (!*svBit) {}
  // do something when temp=1;
}
```

I understand Verilator does not support calling time consuming exported tasks from imported DPI-C functions/tasks. Is there any way I can detect this signal change or write code in a different way?

abarajithan11, 2023-08-27:
I tried the following as well

```
// top_tb.sv

module_top_tb;
  bit signal= 0;
  import "DPI-C" function void run_c ();

  export "DPI-C" function get_signal;
  function get_signal (inout t);
    t = signal;
  endfunction

  initial #1000ns signal=1;
  initial run_c();
endmodule
```
```
// run_c.c

#include <svdpi.h>
extern "C" get_signal (svBit* signal);
extern "C" void run_c () {
  svBit* signal;
  *signal = 0;

  // do something when signal=0;

  while (!*signal) { get_signal (signal); }

  // do something when signal=1;
}
```

wsnyder, 2023-08-27:
You can't have a while() loop in your C code even if using DPI (unless doing something complicated with coroutines.)

Normally this would be handled by having the simulator call your code as needed, it could be on every edge of a signal (always_comb) but that is slow and rarely needed, more likely you can call it on e.g. a clock edge and pass in the signal of interest. Note also it's faster to pass in the signal as an argument, rather than an e.g. get_signal().

     always_ff @(negedge clk)  run_c(signal);

Then your run_c needs to instead of a for loop be e.g. a finite state machine.

      extern "C" void run_c (bool signal) {
          static int fsm = 0;   // Better, use enum 0=uninit, 1 = first action, etc
          if (fsm == 0) {
                fsm = 1;
                // initial stuff
          } else {
                if (signal != last_signal) {
                      // do whatever, perhaps update fsm to change to new state
                }
          }

closed 2023-08-27 by wsnyder

donlon, 2023-08-28:
You are defining an imported DPI-C function, however a function cannot have any delay and if it enters a loop the simulator would be blocked until it's returned (since it's actucally a coroutine rather than multi-threading). But you can try contexted DPI-C task instead, and put the time-consuming part into the SV code, likes

```SystemVerilog
// top_tb.sv
module top_tb;
    bit signal = 0;

    import "DPI-C" context task run_c();
    export "DPI-C" sv_wait_signal = task wait_signal;

    task wait_signal();
        wait (signal == 1);
    endtask

    initial begin
        #1000ns
        $display("signal = 1");
        signal = 1;
    end

    initial run_c();

endmodule
```

```C++
// run_c.c
#include <svdpi.h>
#include <stdio.h>

#ifdef __cplusplus
extern "C" {
#endif

extern void sv_wait_signal();

int run_c() {
    sv_wait_signal();
    printf("signal is now changed to 1\n");
    return 0;
}

#ifdef __cplusplus
}
#endif
```

It's supported by common commercial simulators, however it seems that Verilator has some bug about it. The output order is incorrect. Maybe it will be fixed in the future.