verilator/verilator #4443: A vicious bug in simulation that wave dump value is different to actually value

tastynoob, 2023-08-27:
verilator version: Verilator 5.015 devel rev v5.014-31-g63db60f64
PS: this bug not happened at 5.012

code:
```verilog
typedef struct {
    logic a;
} Data_t;
Data_t buffer[16];
........
// bug code
if (i_vld) begin
    if (buffer[idx].a) begin : bug_block // BUG: I set all buffer[x].a to constant 0
        // BUG: but it is still executed
        new_idx <= (idx == 15) ? 0 : idx + 1;
        $display("a %b : %h", buffer[idx].a, idx);
       // LOG: buffer[idx].a sometimes is 0 and sometimes is 1
       // but in the dumped wave， buffer[idx].a is always 0
    end
    else begin
        new_idx <= idx;
    end
end
........
```

I'm not sure what the cause of this bug
at verilator 5.012 has not this bug and bug_block is not be executed
(ps: verilator 5.012 i also find a bug that "lvalue required as left operand of assignment " in generated c++ code)
(what they have in common is member variables involving struct)

tastynoob, 2023-08-27:
i have push the repo which has bug: https://github.com/tastynoob/aura-core.git
command:
```shell
make aura USE_TRACE=1 -j8
./build/aura -f firmware.bin -e t2000
```

wsnyder, 2023-08-27:
I don't see the code you mention in the post in the repo, and anyways the repo is more complicated than something we'd usually debug for you. Ideally we'd want this reduced to a test that is minimal (as in shows the problem and can't be further simplified), only needs a single verilator command line invocation (versus your own makefiles), and self-checks pass/fails.  See the docs/internals.rst for more guidance on writing such tests.

Other ideas to debug it yourself:
1. Check the sensitivity list acting on the variable.  You didn't include that above.  If it's not a simple posedge/negedge perhaps your code has a race and both answers are legal by IEEE?
2. Using git bisect on verilator, what is the exact verilator change that broke the test?
3. Does turning off any of the various verilator optimizations "-fno*" avoid the problem?

tastynoob, 2023-08-27:
sorry, this code was simplified
(actual bug code is https://github.com/tastynoob/aura-core/blob/56b93237bf9f1d82928e8563dea44f3ebde45e1a/rtl/cpu/frontend/bp/FTQ.sv#L133C70-L133C70)
this is a single clock block, and buffer[idx].a is always 0 in wave
and i can not sure what the cause the bug, because it was work perfectly at verilator 5.012
i also tried many random seeds
i am try to use -O0 argument, but it will throw one interface error and i can not find way to solute it
i also checking the c++ code which was generated by this bug sv code and it looks very normal
logically if the bug _block was executed, the buffer[idx].a should be 1 and $display should output 1 too
but what is puzzling is buffer[idx].a always 0 in wave and $display output sometimes is 0
I've been stuck with this bug for two days
In the end I found verilator 5.012 has no this bug
(actually it has other bugs in 5.012 but easy to solve)

wsnyder, 2023-09-26:
Closing until get small test case.

closed 2023-09-26 by wsnyder