verilator/verilator #4450: Adding source annotation in Generated C++ Sources

AmeyaVS, 2023-08-30:
What would you like added/supported?
I would like to know on how can I add source annotations on the generated C++ Source files mapping it back to Verilog and SystemVerilog RTL sources.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
I can possibly provide an example but currently it requires manually editing the verilated source files.

May we assist you in trying to fix this in Verilator yourself?
General direction on how to proceed with this should be fine, any other discussion platform to discuss on this issue is also fine.

wsnyder, 2023-08-30:
Can you please describe your use case?

Wat you mean by "annotation" - what do you expect this to do?

Note with optimizations there's not a one-to-one correspondence between SV code and C++.

AmeyaVS, 2023-08-30:
I am thinking about following:

- Connecting to the verilated simulation executable to gdb.
- Able to observe simulation state of various Verilog/SV Sources from GDB (This one would require knowing about the internals of verilator objects to observe state in GDB).

Basically an additional command line option to enable this `-annote-verilated-cpp`, .

Referring to following [GCC Line Control Macro](https://gcc.gnu.org/onlinedocs/cpp/Line-Control.html#Line-Control):

> bison or any such program can arrange this by writing ‘#line’ directives into the output file. ‘#line’ is a directive that specifies the original line number and source file name for subsequent input in the current preprocessor input file.

> Note with optimizations there's not a one-to-one correspondence between SV code and C++.

Though I agree, but would it be possible to get a minimal source annotations to get started?

closed 2023-08-30 by wsnyder

AmeyaVSingh, 2023-08-31:
> (note this is already done for `systemc inserted content).

Can you the relevant source reference?

wsnyder, 2023-08-31:
V3EmitCFunc.h 554

```
    void visit(AstComment* nodep) override {
        string at;
        if (nodep->showAt()) {
            at = " at " + nodep->fileline()->ascii();
            // If protecting, passthru less information about the design
            if (!v3Global.opt.protectIds()) return;
        }
        if (!(nodep->protect() && v3Global.opt.protectIds())) {
            putsDecoration(string{"// "} + nodep->name() + at + "\n");
        }
```

Carotti, 2024-06-24:
@wsnyder it sounds like your concern around the usefulness of the `#line` directive is related to the accuracy (e.g. for meaningful debug with `gdb`) - is that fair?

There's another use-case for the same feature, which would be for profiling the simulation performance of larger designs. For performance profiling, even being able to associate a given verilated line with an SV file/module could be useful.

wsnyder, 2024-06-30:
>For performance profiling, even being able to associate a given verilated line with an SV file/module could be useful.

Note for profiling we currently support process level, see `--prof-cfuncs`.