verilator/verilator #4451: Systemverilog repeat loop (counter) lifetime error

jwoutersymatra, 2023-08-31:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

module repeat_lifetime_bug;
initial begin
  fork
    begin
      repeat(5) begin
        $display("iteration");
      end
      //for(int i=0; i<5; i++) begin
      //  $display("iteration");
      //end
    end
  join_any
  $finish;
end
endmodule

What 'verilator --version' are you using?  Did you try it with the git master version?
5.014

What OS and distribution are you using?
Ubuntu under WSL2

May we assist you in trying to fix this in Verilator yourself?
Seems related to previous issue posted and fixed yesterday

wsnyder, 2023-08-31:
@kboronski-ant I fixed the assertion with this:

```
diff --git a/src/V3LinkJump.cpp b/src/V3LinkJump.cpp
--- a/src/V3LinkJump.cpp
+++ b/src/V3LinkJump.cpp
@@ -196,6 +196,7 @@ private:
         // Spec says value is integral, if negative is ignored
         AstVar* const varp
             = new AstVar{nodep->fileline(), VVarType::BLOCKTEMP, name, nodep->findSigned32DType()};
+        varp->lifetime(VLifetime::AUTOMATIC);
         varp->usedLoopIdx(true);
         m_modp->addStmtsp(varp);
         AstNode* initsp = new AstAssign{
```

But there's still a LIFETIME error - should this get suppressed for temporaries?

Maybe we should also be ignoring the lifetime isNone assertion for temporaries?  If not we probably want V3Broken to check that lifetime is set once V3Width completes, otherwise we'll have more of these sort of bugs.

Not sure your thoughts here - mind making a pull to solve it however's appropriate?

wsnyder, 2023-08-31:
I'll merge the patch once the regressions pass, as seems harmless at worst.

kbieganski, 2023-10-23:
This patch:
```diff
diff --git a/src/V3LinkJump.cpp b/src/V3LinkJump.cpp
index e777ac164..5687fa626 100644
--- a/src/V3LinkJump.cpp
+++ b/src/V3LinkJump.cpp

         AstVar* const varp
             = new AstVar{nodep->fileline(), VVarType::BLOCKTEMP, name, nodep->findSigned32DType()};
         varp->lifetime(VLifetime::AUTOMATIC);
+        varp->funcLocal(true);
         varp->usedLoopIdx(true);
-        m_modp->addStmtsp(varp);
+        nodep->addHereThisAsNext(varp);
         AstNode* initsp = new AstAssign{
             nodep->fileline(), new AstVarRef{nodep->fileline(), varp, VAccess::WRITE}, countp};
         AstNode* const decp = new AstAssign{
```
fixes it, but breaks other tests.

wsnyder, 2025-03-31:
This got fixed somewhere along the line, passes in master.

closed 2025-03-31 by wsnyder