verilator/verilator #4452: Hashing tests sometimes fails

kamilrakoczy, 2023-08-31:
I've noticed that hashing tests:
```
test_regress/t/t_inst_long_bad.pl
test_regress/t/t_mod_longname.pl
test_regress/t/t_protect_ids_key.pl
```
sometimes fails:
```
F1: %Error: obj_vlt/t_inst_long_bad/t_inst_long.v:4:3: Cannot find file containing module: 'long_long_long_long_long_long_lo__VhshBv8BYAqnzRahsBWLei19Z4Q6TLrnAouzJIq4TAnb'
F2: %Error: obj_vlt/t_inst_long_bad/t_inst_long.v:4:3: Cannot find file containing module: 'long_long_long_long_long_long_lo__Vhsh1JZCXQVBM1QiASYlLmgTuAXYyUr7VAbJYwVHfiAD'
```
```
%Warning-MULTITOP: t/t_mod_longname.v:28:8: Multiple top level modules
                                          : ... Suggest see manual; fix the duplicates, or use --top-module to select top.
                   ... For warning description see https://verilator.org/warn/MULTITOP?v=5.015
                   ... Use "/* verilator lint_off MULTITOP */" and lint_on around source to disable this message.
                                          : ... Top module 't'
   13 | module t ();
      |        ^
                                          : ... Top module 'modlongnameiuqyrewewriqyewroiquy__VhshPUGhVdBpLQxpA8Ifjh1JyH2zuvWn4rG8CUprAXWw'
%Error: t/t_mod_longname.v:22:4: Cannot find file containing module: 'modlongnameiuqyrewewriqyewroiquy__Vhsh69M9sSPVBT4bNrlWoJWTm6sZeAQ9AQIA826AYpQk'
        ... Unsupported: Name is longer than 127 characters; automatic file lookup not supported.
        ... Suggest putting filename with this module/package onto command line instead.
 ```
 ```
 %Warning: vlt/t_protect_ids_key: Line 5 miscompares; obj_vlt/t_protect_ids_key/Vt_protect_ids_key__idmap.xml != t/t_protect_ids_key.out
F1:   <map from="PSh9cB" to="TOP__t__DOT__secret_inst"/>
F2:   <map from="PSlhR1" to="TOP__t__DOT__secret_inst"/>
```

I've also noticed, that adding ``--debug`` flag to the test makes the test pass.

I've tested multiple verilator versions including current master and commit from 3 months ago that I'm sure all tests were passing (for me) locally, so I think verilator version doesn't matter .

I can only see this behavior only locally (in CI this tests passes), but I confirmed that they also fails for other people using different OS/distributions:
```
Arch Linux, kernel 6.4.12
Arch Linux, kernel 6.3.1
Fedora, kernel 6.3.11
```

but I also confirmed that for ``Debian 11`` this tests are still passing.

donlon, 2023-08-31:
I also noticed that these tests always fail in my Manjaro Linux. Now it seems that this is not an isolated case...

After debugging, I found out that it's because `VHashSha256` generates unstable hashes that make the design fail to link, and it's due to reusing a variable from the previous scope.

I'm creating a PR to fix it.

wsnyder, 2023-08-31:
Thanks for the fix!

closed 2023-08-31 by wsnyder

kamilrakoczy, 2023-09-01:
I can confirm, that this fix also works for me. Thank you for debugging and quick fix.