verilator/verilator #4457: wait(event) causes compile error rather than parse error

jwoutersymatra, 2023-09-06:
Thanks for taking the time to report this.

Compile error "error: invalid cast from type 'VlEvent' to type 'IData' {aka 'unsigned int'}

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

module wait_event_bug;

event e_my_event;
logic do_something;

initial begin
  #(1us);
  wait(e_my_event);
  // wait(e_my_event.trigered);
  do_something = 1;
end

initial begin
  #1ms;
  $finish;
end

endmodule

What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.014 2023-06-06

What OS and distribution are you using?

Ubuntu under WSL

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2023-09-07:
The cast error is easily resolved with the below patch, but that then still compares the VlEvents. I'm not sure what's proper behavior here, @gezalore perhaps?

```
diff --git a/src/V3Cast.cpp b/src/V3Cast.cpp
--- a/src/V3Cast.cpp
+++ b/src/V3Cast.cpp
@@ -186,7 +186,7 @@ private:
             && !VN_IS(backp, NodeCCall) && !VN_IS(backp, CMethodHard) && !VN_IS(backp, SFormatF)
             && !VN_IS(backp, ArraySel) && !VN_IS(backp, StructSel) && !VN_IS(backp, RedXor)
             && (nodep->varp()->basicp() && !nodep->varp()->basicp()->isTriggerVec()
-                && !nodep->varp()->basicp()->isForkSync()
+                && !nodep->varp()->basicp()->isEvent() && !nodep->varp()->basicp()->isForkSync()
                 && !nodep->varp()->basicp()->isProcessRef())
             && backp->width() && castSize(nodep) != castSize(nodep->varp())) {
             // Cast vars to IData first, else below has upper bits wrongly set
```

gezalore, 2023-09-08:
An event has no value, so passing an expression of event type to `wait` is an error, and most tools report it as such. `wait(event.triggered)` should work and is discussed explicitly in the standard in 15.5.3 Persistent trigger: triggered built-in method. I would expect the current implementation should just work with `wait(event.triggered)`, though I have not tried it.

wsnyder, 2023-09-10:
With `wait(event.triggered)` passes.  (Note the commented out line misspelled triggered, I fixed that above.)

closed 2024-11-05 by wsnyder