verilator/verilator #4462: Add in trace API even if tracing disabled.

phelter, 2023-09-07:
Would like to have the trace API be added (and do nothing) when verilated code is generated without `--trace` or `--trace-fst`

The reason for this change is to allow the same testbench infrastructure for both tracing and no tracing.  I use a C++ template [CRTP](https://en.wikipedia.org/wiki/Curiously_recurring_template_pattern) that injects the dut or testbench (verilated top level code) into the the template.  it does a bunch of common functionality such as creating and opening  filenames for the coverage and trace files, and a few other things related to creating a test framework.

When I disable trace verilating the verilog - the resulting toplevel class method `trace()` no longer exists and then causes the templated framework to not compile.

Would it be possible to ensure under all scenarios of verilate commands that the entire API gets defined for each of the modules, but when a feature is disabled, - eg trace - those specific commands do nothing?

There is already a check for context->traceEverOn() inside the VerilatedContext - which seems to be this runtime/programmatic method of doing this.

Right now my workaround is to always enable `--trace` in the command line and set `m_context->traceEverOn(false)` in the m_context.

It is good practice to define one and only one APi which is consistent regardless of verilated command options, and instead of removing the members or methods based on verilate command arguments - make the members or methods empty (do nothing).

Looking at the resulting code, the only benefit of removing the member function is a missing entry in a vtable, and a little bit less code.

Much appreciated.

wsnyder, 2023-09-08:
This seems a reasonable idea, however instead of trace() being a nop I'd think it should flag an error, to help if someone is trying to debug why they aren't getting traces.

Might you prepare a pull request to do this?

closed 2023-09-26 by wsnyder

phelter, 2023-09-27:
Thanks for this @wsnyder