verilator/verilator #4471: Fix variable not declared with fork local

wsnyder, 2023-09-10:
This test, I will commit shortly (based off iv-test's br_gh4a):

```
module t();
  initial fork
     reg i;
     i = 1'b1;
     if (i != 1'b1) $stop;
     $write("*-* All Finished *-*\n");
     $finish;
  join
endmodule
```

Crashed, but I'm committing a "if (initsp()" to V3Fork to fix it.  It now Verilates but at compile time gives

```
Vt_fork_initial_t__DepSet_h667e1af4__0.cpp: In function 'VlCoroutine Vt_fork_initial_t___eval_initial__TOP__t__0(Vt_fork_initial_t*)':
Vt_fork_initial_t__DepSet_h667e1af4__0.cpp:19:5: error: '_Vwrapped_0x555556e4ee00__DOT____VDynScope_0x555556e4ee00' was not declared in this scope
   19 |     _Vwrapped_0x555556e4ee00__DOT____VDynScope_0x555556e4ee00
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

@kboronski-ant perhaps you could look if you get a chance?

wsnyder, 2023-09-10:
Test pushed as t_fork_initial.v.  Edit t_fork_initial.pl as described there to see this error.

kboronski-ant, 2023-10-12:
Ok, I see what's happening here. `initial` block has a static lifetime which results in the dynscope object getting moved into vlSelf class. However, not everything gets moved like that in static context. I added an initial block that operated on an integer and it stayed inside of function's body. Maybe just objects get moved like that, idk. If I knew for certain, this could be fixed by adding a `vlSelf->` when referencing the dynscopes when appropriate. An even better solution would be to ensure that the referenced variables are moved to the class, instead of creating a `DynScope`, but this would be harder to achieve.