verilator/verilator #4472: Corrupted .vcd file when assigning -1 to port with bitsize 1

miree, 2023-09-10:
### Description 
Assingment of -1 in the C++ testbench to a port that was declared as [0:0], i.e. with bitsize 1 results in corrupted .vcd file.

The described behavior appears only with ports of bisize 1. With ports of bitsize >1 (e.g. [4:0]) this doesn't happen. Instead all bits are set to 1 (b11111 appears in the waveform.vcd) when the integer value -1 is assigned to the port in a C++ testbench, which is I guess the intended behavior.

### Version
This behavior is observed in Verilator 5.014, I've not attempted to reproduce this with different versions.
### Reproduce
bug.v
```Verilog
module bug (
	input clk,
	input reg[0:0] i,
	output reg[0:0] o  
);
assign o = i;
endmodule
```
bug_tb.cpp
```C++
#include "verilated.h"
#include "Vbug.h"
#include "verilated_vcd_c.h"
#define MAX_SIM_TIME 20
vluint64_t sim_time = 0;

int main(int argc, char** argv, char**) {
    Verilated::debug(0);
    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
    contextp->traceEverOn(true);
    contextp->commandArgs(argc, argv);

    const std::unique_ptr<Vbug> topp{new Vbug{contextp.get()}};
    VerilatedVcdC *m_trace = new VerilatedVcdC;
    topp->trace(m_trace, 5);
    m_trace->open("waveform.vcd");

    while (sim_time < MAX_SIM_TIME) {
        topp->clk ^= 1;
        if (sim_time == MAX_SIM_TIME/2) topp->i = -1; // <=== ASSIGN MINUS ONE HERE  ======================
        topp->eval();
        m_trace->dump(sim_time);
        ++sim_time;
    }
    m_trace->close();
    if (!contextp->gotFinish()) {
        VL_DEBUG_IF(VL_PRINTF("+ Exiting without $finish; no events left\n"););
    }
    topp->final();
    return 0;
}

```
comile and execute resutls in a corrupted **waveform.vcd**
```bash
[verilator_bug]$ verilator --trace -cc --top-module bug bug.v --exe bug_tb.cpp
[verilator_bug]$ make -C obj_dir/ -f Vbug.mk
[verilator_bug]$ obj_dir/Vbug
[verilator_bug]$ cat waveform.vcd 
$version Generated by VerilatedVcd $end
$timescale 1ps $end

 $scope module TOP $end
  $var wire  1 # clk $end
  $var wire  1 $ i [0:0] $end
  $var wire  1 % o [0:0] $end
  $scope module bug $end
   $var wire  1 # clk $end
   $var wire  1 $ i [0:0] $end
   $var wire  1 % o [0:0] $end
  $upscope $end
 $upscope $end
$enddefinitions $end

#0
1#
0$
0%
#1
0#
#2
1#
#3
0#
#4
1#
#5
0#
#6
1#
#7
0#
#8
1#
#9
0#
#10
1#
�$   <====== INVALID CHARACTER HERE ======================
�%   <====== INVALID CHARACTER HERE ======================
#11
0#
#12
1#
#13
0#
#14
1#
#15
0#
#16
1#
#17
0#
#18
1#
#19
0#
```

wsnyder, 2023-09-10:
It is illegal to set bits outside the input width of the signal (e.g. -1 sets all bits which is illegal as only bit 0 may be set).  If you compile with VL_DEBUG you will get an assertion "Testbench C set input '...' to value that overflows what the signal's width can fit when this is done.

closed 2023-09-10 by wsnyder

miree, 2023-09-10:
Ok, thanks for the hint. Compiling with VL_DEBUG indeed stops at runtime, even for lager port sizes.

However, when assigning a positive value that exceeds the bitsize of the port, verilator outputs the following warning during compilation which suggests that assignment of -1 is a valid option:
```bash
../bug_tb.cpp: In function ‘int main(int, char**, char**)’:
../bug_tb.cpp:20:51: warning: unsigned conversion from ‘int’ to ‘CData’ {aka ‘unsigned char’} changes value from ‘65535’ to ‘255’ [-Woverflow]
   20 |         if (sim_time == MAX_SIM_TIME/2) topp->i = 0xffff; // <=== ASSIGN MINUS ONE HERE
      |                                                   ^~~~~~
```

wsnyder, 2023-09-10:
Note verilator cannot analyze your C++ wrapper. That's a gcc warning, not Verilator's, also indicating trouble.

miree, 2023-09-10:
Sorry, you're right. I got confused by the comment in my own code.