verilator/verilator #4480: Segmentation fault on commit e24197f

kuopinghsu, 2023-09-11:
* What 'verilator --version' are you using?  Did you try it with the git master version?

Segmentation fault on $fwrite function on commit e24197f, no issue on commit 63db60f. The issue still exist on git master version.

* What OS and distribution are you using?

WSL2 Ubuntu 23.10 + gcc v11 & v12
MacOS clang v14.0.3

The problem can be replicated with both envs. MacOS always stops at segmentation fault. In Ubuntu, running multiple times, segmentation fault occurs about 70% of the time and passes 30% of the time.

* How to replicate

I can't reproduce this issue with a simple test case. Attached [verilate_e24197f.tar.bz](https://drive.google.com/file/d/1vNkrQAC3P10BDSK9pajQaOPT4_ud26QZ/view?usp=drive_link) is a complex test case. Use "make failed" to run the failed case, and use "make pass" to run the passed case.

Search the source tree for the keyword "FAILED_CASE" to see the differences to test. An error occurred on the $fwrite function.

wsnyder, 2023-09-11:
Can you run with --debug --gdbbt and post the backtrace?  This should point to the relevant source line - perhaps you can also look at the sources and propose a pull to fix it?

kuopinghsu, 2023-09-12:
After adding "--debug --gdbbt", the failure rate becomes 50%. Invoke gdb to launch it, it becomes always passes. This is sensitive to compilation options. Unfortunately, I am not able to get the backtrack. Still looking for how to debug it.

kuopinghsu, 2023-09-12:
gdb backtrace

```
_vl_vsformat (output="000f", formatp=formatp@entry=0xaaaaaaab8ad4 "%02x%02x%02x%02x\n", ap=...) at /usr/share/verilator/include/verilated.cpp:901
901	                    ld = lwp[0];
(gdb) where
#0  _vl_vsformat (output="000f", formatp=formatp@entry=0xaaaaaaab8ad4 "%02x%02x%02x%02x\n", ap=...)
    at /usr/share/verilator/include/verilated.cpp:901
#1  0x0000aaaaaaab08fc in VL_FWRITEF (fpi=fpi@entry=2147483692, 
    formatp=formatp@entry=0xaaaaaaab8ad4 "%02x%02x%02x%02x\n") at /usr/share/verilator/include/verilated.cpp:1595
#2  0x0000aaaaaaab5bfc in Vriscv___024root___nba_sequent__TOP__1 (vlSelf=vlSelf@entry=0xfffff77a5100)
    at /home/ubuntu/test/sim_cc/Vriscv___024root__DepSet_h309ef4e5__0.cpp:877
#3  0x0000aaaaaaab6dd4 in Vriscv___024root___eval_nba (vlSelf=vlSelf@entry=0xfffff77a5100)
    at /home/ubuntu/test/sim_cc/Vriscv___024root__DepSet_h309ef4e5__0.cpp:2362
#4  0x0000aaaaaaab6f48 in Vriscv___024root___eval (vlSelf=0xfffff77a5100)
    at /home/ubuntu/test/sim_cc/Vriscv___024root__DepSet_h309ef4e5__0.cpp:2423
#5  0x0000aaaaaaab6fe4 in Vriscv::eval_step (this=0xaaaaaaae2180) at /home/ubuntu/test/sim_cc/Vriscv.cpp:63
#6  0x0000aaaaaaaa5928 in Vriscv::eval (this=0xaaaaaaae2180) at ./Vriscv.h:54
#7  main (argc=<optimized out>, argv=<optimized out>) at ../sim_main.cpp:37
```

wsnyder, 2023-09-12:
The error suggests that there's something misformed in the print, e.g. accessing a word or otherwise past the end of a wide vector. Most likely this relates to some bad input Verilog Verilator isn't properly warning about.

What is the verilog code corresponding to this print?  (If not obvious look at the --debug created .tree file, find the .tree line that matches the .cpp statement, and that .tree line will have an e.g. "{a23}" on it which means file "a" line number 23.

kuopinghsu, 2023-09-14:
This is a simple data write on testbench.v line 212.

```
for (i = `TOP.regs[REG_A0]; i < `TOP.regs[REG_A1]; i = i + 4) begin
    $fdisplay(dump, "%02x%02x%02x%02x", dmem.getb(i - IRAMSIZE + 3),
                                        dmem.getb(i - IRAMSIZE + 2),
                                        dmem.getb(i - IRAMSIZE + 1),
                                        dmem.getb(i - IRAMSIZE + 0));
end
```

And getb() is defined as a function returning type [7:0].

```
module dmem(
...);

function [7:0] getb;
    input [31:0] address;
begin
    if (address[31:ADDRW+2] != 0) begin
        $display("Address %08x out of range", address);
    end

    case(address[1:0])
        0: getb = ram[address[ADDRW+1: 2]][8*0+7:8*0];
        1: getb = ram[address[ADDRW+1: 2]][8*1+7:8*1];
        2: getb = ram[address[ADDRW+1: 2]][8*2+7:8*2];
        3: getb = ram[address[ADDRW+1: 2]][8*3+7:8*3];
    endcase
end
endfunction

endmodule
```

This error only occurs on commit e24197f and do not relate to the $fwrite() function.
Removing the two dmem.getb() function calls, as shown below, does not trigger the problem.

```
    $fdisplay(dump, "%02x%02x", dmem.getb(i - IRAMSIZE + 3),
                                dmem.getb(i - IRAMSIZE + 2));
```

wsnyder, 2023-09-15:
Most likely the ram[] access math is exceeding the bounds of the ram.  You can check this by adding a display of the address (you didn't show declaration of 'ram' so don't know if it matches print you already have.)

Verilator should have added bounds checking for this when needed, look at the .cpp code, and --debug .tree file around those ram accesses to see if you can spot the issue.  Alternatively when in gdb you can print 'address' right after break at the error.

kuopinghsu, 2023-09-15:
The RAM size is 1716Kb and a segmentation fault occurs on the first $fdisplay function. The value of (i-IRAMSIZE) is 272. The index is valid, and I've been wrote this code for over two years without issues until a recent commit. These are some corner cases of this problem and I can't find the root cause. I checked the parameters of $fdisplay and it looks correct and does not exceed memory boundaries.

If I modify the getb function and remove the if expression, the segmentation fault is not triggered. Any ideas?

```verilog
module dmem(
...);

function [7:0] getb;
    input [31:0] address;
begin
    // if (address[31:ADDRW+2] != 0) begin
    //     $display("Address %08x out of range", address);
    // end

    case(address[1:0])
        0: getb = ram[address[ADDRW+1: 2]][8*0+7:8*0];
        1: getb = ram[address[ADDRW+1: 2]][8*1+7:8*1];
        2: getb = ram[address[ADDRW+1: 2]][8*2+7:8*2];
        3: getb = ram[address[ADDRW+1: 2]][8*3+7:8*3];
    endcase
end
endfunction

endmodule
```

kuopinghsu, 2023-09-16:
Well, this is difficult for me to debug this issue. I tried to simply test and the test here is simple enough to trigger this issue.

```verilog
module test(
    input clk
);
    integer i;
    integer count = 'd0;

    memory memory();

always @(posedge clk) begin
    if (count < 12)
        count <= count + 1;

    if (count == 10) begin
        for(i=0; i<300; i=i+4) begin
            $display("%02x%02x%02x%02x", memory.getb(i+3), memory.getb(i+2), memory.getb(i+1), memory.getb(i));
        end
    end

    if (count == 11) begin
        $finish(0);
    end
end

endmodule;

module memory;

    localparam SIZE=64*1024;
    localparam ADDRW = $clog2(SIZE/4);

    reg  [31: 0] ram [(SIZE/4)-1: 0];
    integer i;

function [7:0] getb;
    input [31:0] address;
begin
    if (address[31:ADDRW+2] != 0) begin
        $display("Address out of range");
    end

    case(address[1:0])
        0: getb = ram[address[ADDRW+1: 2]][8*0+7:8*0];
        1: getb = ram[address[ADDRW+1: 2]][8*1+7:8*1];
        2: getb = ram[address[ADDRW+1: 2]][8*2+7:8*2];
        3: getb = ram[address[ADDRW+1: 2]][8*3+7:8*3];
    endcase
end
endfunction

initial
begin
    for (i=0; i<SIZE/4; i=i+1) begin
        ram[i] = 32'hdeadbeaf;
    end
end

endmodule
```

As before, remove the if expression in the getb function and the error disappears.

The error occurred between commits e24197f and 63db60f. Commit 63db60f is fine, but commit e24197f has a segmentation fault on the $display statement in the module test.

In commit 63db60f, the generated $display statement is straightforward.

```c
            VL_WRITEF("%02x%02x%02x%02x\n",8,vlSelf->__Vfunc_getb__0__Vfuncout,
                      8,(IData)(vlSelf->__Vfunc_getb__1__Vfuncout),
                      8,vlSelf->__Vfunc_getb__2__Vfuncout,
                      8,(IData)(vlSelf->__Vfunc_getb__3__Vfuncout));
```

In commit e24197f, It looks like not enough arguments are generated for the VL_WRITEF() function.

wsnyder, 2023-09-16:
Great that isolation was just what was needed. I see what was wrong, the commit e24197f exposed a latent bug where a $display under another $display corrupted internal state, causing the dropped arguments. Fixed.

wsnyder, 2023-09-16:
Please confirm it works in your larger test case.

closed 2023-09-16 by wsnyder

kuopinghsu, 2023-09-16:
Thank you. It works in my test case in MacOS Sonoma 14.0 and Ubuntu 22.04 env.