verilator/verilator #4482: More helpful error message on Enum <-> Instantiation name collision

meetowl, 2023-09-11:
Hi all, 
I've been going through a book which creates a basic microprocessor, and after much debugging, I figured out that it called an instantiation the same name as an enum element.

The current error message hints an internal error, specifically
```
%Error: Internal Error: CPU.v:2:6: ../V3Ast.h:2098: AstNode is not of expected type, but instead has type 'CELL'
    2 |    S s1 (clk);
      |      ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
is not clear, and it took me a lot of digging to figure out why it was popping up.

A cut-down reproducible version is this, not sure how else to fit test_regress:
```
module CPU (input logic clk);
   S s1 (clk);
endmodule // CPU

// Sequences operations in a state machine
module S (input clk);
   enum {s0, s1} state;
   always_ff @(clk)
     begin
        $write("*-* All Finished *-*\n");
        $finish;
     end
endmodule // s
```

`verilator --version` = 5.014

I am using Manjaro Linux.

I wouldn't know where to begin with fixing this, but if you could point me to where I can create more helpful error messages I'll have a crack at fixing this.

Thank you all!

donlon, 2023-09-11:
The error is happening because the name of the instantiated module `s1` is the same as an enum item inside the module. The backtrace in debugger shows that the error happens [here](https://github.com/verilator/verilator/blob/e77d847671ba3b6a5c088a82da1eb1809d5dd665/src/V3LinkDot.cpp#L1354). This line of code reinterpret `foundp->nodep()` pointer to AstEnumItem, but it's not an AstEnumItem actually. It's an AstCell which is obtained from `foundp`, which has been assigned by `m_modSymp` above this line. By removing these lines the error seems to be fixed, but I'm not sure why it's doing this... @wsnyder

The `findIdFallback` function is used when it's finding for name duplication, but instead of using `findIdFlat`, this function allow searching for symbols in the upper level if no symbol is matched in the current scope. Verilator also throws a VARHIDDEN warning (although it's a `styleError` and it's default off) if the enum's name conflicts to enums in the upper level. I doubt whether this check is necessary.

For AstVar, symbol insertion [procedure](https://github.com/verilator/verilator/blob/e77d847671ba3b6a5c088a82da1eb1809d5dd665/src/V3LinkDot.cpp#L1200) is basically the same, but it uses `VN_CAST`, so there won't be such error.

wsnyder, 2023-09-12:
Thanks for the good test & analysis. Instead of VN_AS it should have been VN_CAST.  Also the warning message resulting was pointing at the wrong source code text. Both fixed.

closed 2023-09-12 by wsnyder