verilator/verilator #4483: Parameter without default value (--lint-only)

wespiard, 2023-09-11:
When running Verilator with the `--lint-only` flag, I am getting the following error if I have a parameter without a default value:
> Parameter without initial value is never given value (IEEE 1800-2017 6.20.1)

See the following example from the `test_regress/t` directory:
https://github.com/verilator/verilator/blob/master/test_regress/t/t_param_noval_bad.v
```
module t #(parameter P);
   generate
      var j;
      for (j=0; P; j++)
        initial begin end
   endgenerate
endmodule
```

In this case, the parameter `P` not having a default value returns the error in question, when just linting the file.

The IEEE Std 1800-2017 LRM in section 6.20.1 states the following:
> In the declaration of a parameter in a parameter port list, the specification for its default value may be
omitted, in which case the parameter shall have no default value.

I understand that it is required for and instantiated module to have a parameter value specified, but I don't think it should be a requirement for every parameter to have a default value.

Only tried with the following version of Verilator, as I think this is more of a question of rules, not a code bug.
`Verilator 5.014 2023-08-06 rev v5.014-35-ge6b0bdd4d`

What OS and distribution are you using?
RHEL 8.8

May we assist you in trying to fix this in Verilator yourself?
Sure, but I have never looked at the Verilator code base before. 

Is the solution to disable this check in the `--lint-only` mode?

wsnyder, 2023-09-11:
A value is required either as a default in the definition, or when instantiated. 

"If no default value is specified for a parameter of a design element, then an overriding parameter value shall be specified in every instantiation of that design element (see 23.10).

I think you're arguing that it's useful to be able to lint an incomplete design outside of what IEEE says.  That's reasonable, but a simulator has no idea what to use for the parameter value - it can't just make something up. So there's no fix possible here, instead add a default value for the parameter that makes sense for linting purposes (which is what other sites do BTW).

closed 2023-09-11 by wsnyder

wespiard, 2023-09-11:
I see. I understand from a simulation point of view; but I assumed that for the purposes of linting there could be some option.

Is the typical flow when linting with Verilator to actually pass all files in the project as if I was compiling for simulation, including all parameter values? I'd rather not add fake default values and then have to add checks such that they aren't actually left with the fake values. 

I like the failsafe of a simulator or synthesis tool throwing an error that a parameter is not defined, but not when I'm just linting.

wsnyder, 2023-09-12:
Well you could hack your code with a default only when linting, e.g.

   parameter foo `ifdef LINT = 1 `endif

All the library modules we write where I work we always make sure the module works with default parameters, so we can lint it.