verilator/verilator #4484: How to correctly export SystemVerilog function/task via DPI to C?

abarajithan11, 2023-09-11:
I have these functions in `file_1.sv`

```
  bit [7 :0] y_mem  [0:1000-1];
  int addr_8;
  logic signed [3:0][7:0] data_8;

  export "DPI-C" function write_y;
  export "DPI-C" function read_y;

  function void write_y (input longint addr, input int data);
    data_8 = data;
    for (int i=0; i<4; i++) 
      y_mem[addr + i] = data_8[i];
  endfunction

  function int read_y (input longint addr);
    for (int i=0; i<4; i++) 
      data_8[i] = y_mem[addr + i];
    read_y = data_8;
  endfunction
```

In `file2.c`, I import them as:
```
extern "C" void write_y (int * addr, int data);
extern "C" int  read_y  (int * addr);
```

And I get the error:
```
%Error: unknown:0: Testbench C called 'write_y' but scope wasn't set, perhaps due to dpi import call without 'context', or missing svSetScope. See IEEE 1800-2017 35.5.3.
```

I tried setting the scope:
```
svSetScope (svGetScopeFromName ("file_1.sv"));
```

and I get the error:
```
../../c/file2.c:15:13: error: expected constructor, destructor, or type conversion before ‘(’ token
   15 |   svSetScope(svGetScopeFromName("file_1.sv"));
      |             ^
```

I tried changing the function to task, same error. 

How do I correctly export an SV function/task?

wsnyder, 2023-09-11:
>../../c/file2.c:15:13: error: expected constructor, destructor, or type conversion before ‘(’ token

This error suggests you aren't putting that code inside a function, you need to.  Or, if you still can't see what's wrong attach the whole C++ file.

Also note svGetScopeFromName will require a scope not a filename.  You might want to look at some examples in the test_regress directory.

Closing as answered but feel free to follow up here.

closed 2023-09-11 by wsnyder

abarajithan11, 2023-09-12:
Here is an example with actual files.

file_1.sv
```
`timescale 1ns/1ps

module file_1;
  bit [7 :0] mem [0:99];
  import "DPI-C" function void test();
  export "DPI-C" function write;
  
  function int write (input int addr, input int data);
      mem[addr] = data;
  endfunction

  initial begin
    test();
    #100ns $finish;
  end
endmodule
```

file_2.c
```
#include <svdpi.h>

extern "C" int write (int addr, int data);

extern "C" void test (){
  write(0, 100);
}

```

command:
```
verilator --binary -j 0 -Wno-fatal --trace --relative-includes --top file_1 file_1.sv file_2.c
```

error now:
```
In file included from Vfile_1__Dpi.cpp:14,
                 from Vfile_1__ALL.cpp:8:
Vfile_1__Dpi.h:20:16: error: conflicting declaration of C function ‘int write(int, int)’
   20 |     extern int write(int addr, int data);
      |                ^~~~~
In file included from /usr/local/share/verilator/include/verilatedos.h:355,
                 from /usr/local/share/verilator/include/verilated.h:42,
                 from Vfile_1.h:11,
                 from Vfile_1.cpp:4,
                 from Vfile_1__ALL.cpp:3:
/usr/include/unistd.h:378:16: note: previous declaration ‘ssize_t write(int, const void*, size_t)’
  378 | extern ssize_t write (int __fd, const void *__buf, size_t __n) __wur
      |                ^~~~~
make: *** [/usr/local/share/verilator/include/verilated.mk:245: Vfile_1__ALL.o] Error 1
rm Vfile_1__ALL.verilator_deplist.tmp
make: Leaving directory '/mnt/d/dnn-engine/test/temp/obj_dir'
%Error: make -C obj_dir -f Vfile_1.mk -j 20 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/bin/verilator_bin --binary -j 0 -Wno-fatal --trace --relative-includes --top file_1 file_1.sv file_2.c
```

wsnyder, 2023-09-12:
>/usr/include/unistd.h:378:16: note: previous declaration ‘ssize_t write

This is saying that write() conflicts with the write() system library function, so you need to use a different function name.

abarajithan11, 2023-09-12:
Ok, I changed the name write to write2 everywhere. And I get the previous error:

```
%Error: unknown:0: Testbench C called 'write2' but scope wasn't set, perhaps due to dpi import call without 'context', or missing svSetScope. See IEEE 1800-2017 35.5.3.
```

if svSetScope should be called within a function, I did as following and I got the same error:

file1.sv
```
`timescale 1ns/1ps

module file_1;
  bit [7 :0] mem [0:99];
  import "DPI-C" function void test();
  export "DPI-C" function write2;
  
  function void write2 (input int addr, input int data);
      mem[addr] = data;
  endfunction

  initial begin
    test();
    #100ns $finish;
  end
endmodule
```

file2.c
```
#include <svdpi.h>
extern "C" void write2 (int addr, int data);
extern "C" void test (){
  svSetScope(svGetScopeFromName("file_1"));
  write2(0, 100);
}
```

Command
```
verilator --binary -j 0 -Wno-fatal --trace --relative-includes --top file_1 file_1.sv file_2.c
./obj_dir/Vfile_1
```

Error:
```
%Error: unknown:0: Testbench C called 'write2' but scope wasn't set, perhaps due to dpi import call without 'context', or missing svSetScope. See IEEE 1800-2017 35.5.3.
Aborting...
Aborted
```

What is the right way of doing this?

wsnyder, 2023-09-12:
1. You need

   import "DPI-C" context function void test();
 
2. You should check if svGetScopeFromName is returning null, which will then tell you that your scope isn't correct.

    svSetScope(svGetScopeFromName("TOP.file_1"));

wsnyder, 2023-09-12:
(or just remove the svSetScope since you want to keep the same scope as it was called from)

abarajithan11, 2023-09-12:
Great. Adding `context` solved it. So, any C function that uses an SV exported function should be imported as `context`.

Final working code:

file_1.sv
```sv
`timescale 1ns/1ps

module file_1;
  bit [7 :0] mem [0:99];
  import "DPI-C" context function void test();
  export "DPI-C" function write2;
  
  function void write2 (input int addr, input int data);
      mem[addr] = data;
  endfunction

  initial begin
    test();
    #100ns $finish;
  end
endmodule
```

file_2.c
```c
#include <svdpi.h>
extern "C" void write2 (int addr, int data);
extern "C" void test (){
  write2(0, 100);
}
```